ArduPilot/ardupilot #2281: Questions about the px4 automatic flight time

luo674526417, 2015-05-12:
I would like to consult that how px4 in automatic flight, suspended task hovering in the sky waiting, then how to continue tasks can be suspended in the current position.thanks!

rmackay9, 2015-05-12:
@luo674526417, this is not really the place to ask questions, this is for recording bugs and enhancement requests.
In any case, you can switch out of AUTO mode into say Loiter or Guided, then when the vehicle is switched back into AUTO it will continue with the mission if the MIS_RESTART parameter is "0".

closed 2015-05-12 by rmackay9

luo674526417, 2015-05-12:
@rmackay9 But from the second switch back to automatic navigation point, how can from the current waypoint to continue

magicrub, 2015-05-12:
@luo674526417

"you can switch out of AUTO mode into say Loiter or Guided, then when the
vehicle is switched back into AUTO it will continue with the mission
if the MIS_RESTART
parameter is "0"
<http://copter.ardupilot.com/wiki/configuration/arducopter-parameters/#mission_restart_when_entering_auto_mode_mis_restart>
."

This can also be done all with auto.
waypoint, then loiter for a time
<http://copter.ardupilot.com/wiki/mission-planning-and-analysis/mission-command-list/#loiter_time>,
then auto some more.