ArduPilot/ardupilot #2282: Crash after Autotune (3.3-rc3+)

hsteinhaus, 2015-05-12:
As reported here (https://groups.google.com/forum/#!topic/drones-discuss/_UCOju4U9kw), an AutoTune run resulted in way to high gains, that caused a crash immediately after takeoff and giving some sharp input. 

Here is the log for the Autotune and the following takeoff with crash: https://www.google.com/url?q=https%3A%2F%2Fwww.dropbox.com%2Fs%2Ftszd0nvu2w92k9v%2F3.BIN%3Fdl%3D0&sa=D&sntz=1&usg=AFQjCNH6SePu6jVasxYimvrnye-HCUuUyw

Before starting the Autotune the quad was flying not too well (gains generally too low), but safe.

rmackay9, 2015-05-13:
Thanks for the report although this is not really the place for raising this issue.  This should go to Leonard on the beta AC3.3 testing thread.

closed 2015-05-13 by rmackay9

hsteinhaus, 2015-05-13:
Hi Randy,

as suspected in the drones-discuss post, I think this crash could be related to the PID voltage scaling and/or the latest AutoTune changes. Unfortunately the post found no attention for a > week now, so I triaged it to here. 

The airframe used for this test is my regular UAVCAN test copter, which has completed AutoTune successfully many times with my AC 3.2.1-UAVCAN branch and early 3.3-dev versions. So one of the changes made to autotune during later 3.3 development (I guess after rc1) obviously is troublesome for at least some airframes. I don't consider this a support problem therefore. 

Holger

rmackay9, 2015-05-13:
@hsteinhaus, I'll ping Leonard to see if he can take a look.

lthall, 2015-05-13:
Unfortunately you needed to have autotune logging turned on in the release you have there. This means there is very little I can see in your logs. We have changed this so autotune logs are always on.

I would wait until the next release (should be this weekend). Do an autotune and then do another test. If you test the tune in the same way straight after autotune has finished you will be able to switch back to your old tune. 

I generally take off and hover at 0.5m to do my first tests of autotune. This lets me drop the copter back onto the ground with little chance of damage.

Let us know how you go and sorry for your crash.

hsteinhaus, 2015-05-13:
I am surprised that the log is not complete, as the LOG_BITMASK was -1, so all possible bits should have been set (that is why the log contains the new raw IMU data).

I will try it again next week with rc4.

lthall, 2015-05-13:
Yeh, it is actually a mission planner problem so I can look at it
. I will get back to you soon!!

rmackay9, 2015-05-13:
It looks to me like it got wobbly after the tune was done and the operator took off in Stabilize mode.  So I don't think it crashed during the AutoTune but instead the AutoTune came up with PID values that were too high and caused oscillations.  As the oscillations grew more severe (the copter was pitching forward 20deg, back 10deg) the operator cut the throttle and it crashed.

With all the logging on it's got *huge* amounts of data in the logs.  In particular the full rate accelerometer logging.  It actually makes them quite hard to work with.

hsteinhaus, 2015-05-13:
Randy, that is exactly what I have described in the linked drones-discuss post. I did not notice any problems directly after autotune while switching between old and new gains. So much bigger was the surprise after landing, disarming (with AutoTune=on to save gains), and taking off again. 

Yes, the full rate logs are a side effect of keeping the LOG_BITMASK at -1 for experimental flights. I usually do some "grep -v" stages on the text logs to filter out unwanted message types before passing them through mavgraph.py - it would be painfully slow otherwise.

lthall, 2015-05-13:
Can you give us a parameter file from after the crash flight?

lthall, 2015-05-13:
Moving to Drones discuss.

hsteinhaus, 2015-05-13:
Just extracted this file via mavparams.py: https://www.dropbox.com/s/5kh8fzcoa0ejdc6/3.param?dl=0