ArduPilot/ardupilot #2285: RC Trim not working Arduplane 3.3.0Beta2

Jman841, 2015-05-12:
I am trying to manually change the trims for the aileron and elevator servos on a plane. When I type a new value and click "Write Parameters" there is no change in the servo trim and when I look at the status menu the RC Output is still the previous value even though the RC Trim value has changed.

I have cycled the power and the RC Trim value saved properly and is correct, however, the RC Output for that channel is still 1514 instead of 1589 that it should be.

Jman841, 2015-05-12:
Imputing a RC Trim does absolutely nothing. I have turned off "TRIM RC AT START" and still changing RC Trim values does absolutely nothing.

Jman841, 2015-05-12:
I loaded Arduplane 3.2.3 and am having the same issue. With the Tx trims at 0, changing trims in arduplane does nothing. Is this a bug or am I just doing something incorrect?

On a different plane in the past I had no issue changing trims with simply changing the RC Trim output in mission planner. I believe this is a bug.

Jman841, 2015-05-12:
If the rcTx is off, the correct value is output to RC1 but with the Tx on, the trims do not do anything. Is this by design or is this a bug? If it is by design, how do I adjust trims so I can keep the trims on the radio centered?

magicrub, 2015-05-12:
what mode are you testing all this in?

Jman841, 2015-05-13:
Manual mode, is that the proper function? If so please close this issue. Is there a way to do sub trims before setting trims on the Tx?

magicrub, 2015-05-13:
I'm not sure if manual mode does the trim, have you tried it with
Stabilized?

Jman841, 2015-05-13:
I will test again today in FBWA, but yesterday it seemed as if it didn't work their either. The only way the trims worked was if the Tx was powered off.

magicrub, 2015-05-13:
And when you power of the Tx you probably go into failsafe and the mode
changes. Or your transmitter outputs it's own failsafe mode.

Jman841, 2015-05-13:
When the Tx is turned off it goes to fail safe but then I change the mode back to manual via the GCS. At this point the trims are proper and the rcoutput in the status page is correct. The Tx is not outputting failsafes as if it were the values would not change to the correct trim after it is powered off and switched back to manual mode via the GCS.

I think the issue is ardupilot always accepts the Tx output and trim is only for other modes or when no Tx output is present. Is there a way to add subtrims so we can trim the plane before moving the tx trims?

magicrub, 2015-05-13:
Do you have any mixing on your transmitter that would make the receive
optout different values in failsafe than normal?

Jman841, 2015-05-13:
No mixing. The Rc calibration shows centering at 1514 which is what all channels output when the tx is connected. When the tx is off the "Rc trim" value I have set for the channels is properly output (ie 1592 for ch1 currently)

Jman841, 2015-05-17:
I confirmed this is definitely an issue. I am using a futaba sbus rx and tx. With pilot inputs turned on for auto modes it completely overrides the auto mode and turns it into FBWA. When we switch into auto mode with a mission loaded the plane just flies flat and and straight until we take back over. We have full control like it is in FBWA even though it is in auto mode and proper waypoints set. I turned off the RC Inputs in auto mode and it does the auto mission. However, we are having serious problems with auto mode currently, but, it is atleast trying to do the mission when RC Inputs during auto modes are turned off.

tridge, 2015-08-14:
I'd need to see a tlog to know what is happening here. Trim is definitely working as designed. Note that the trim values do nothing in manual, as manual directly controls the servos from the transmitter.

skystarUK, 2015-12-11:
I have the same problem with RC TRIM and the Servo limits MIN MAX appearing not to work for Rudder, Aileron and Elevator.  They do work for flaps on RC5.

I have not flown in the air yet with my Pixhawk Bix3 setup, trying to get happy with its movements on the ground first.
Manual mode , no affect at all, upon reading your post, I guess that's normal.
However on the testing I've done with elevator it doesn't seem to be used in FBWA either (or other modes)  perhaps its different once flying, but I am reluctant to try as I am not a great pilot and the elevator atm would be putting me into a dive.

But anyway the main reason I am posting is to query what you said.
Trims don't work in manual by design?
How does that make sense
If I need to set trims using my radio in manual, then I switch to FBWA the trims on my radio will be added to the trims set in pixhawk causing an error, I would then need to recenter trims on radio every time I move out of manual mode?    am I missing something here?

I have an APM arducopter tricopter and that doesn't work like this, the trims are set on the APM for all modes, infact they recommend not using trims on the transmitter other than for setup.

Also you say Manual directly controls the servos from the tx, but does this include ignoring the RC2_MIN  , RC2_MAX  servo limits I set on pixhawk,  sounds like it does, I don't understand why that would make sense.
I set the servo limits to a physical limit, to prevent damaging something.

magicrub, 2016-02-21:
@Jman841  Is this issue still a problem for you?

skystarUK, 2016-02-23:
I still have the issue.
A lot of fiddling around with trims just to see if Pixhawk would respect my settings. I think Flaps and Rudder did although I am not certain about rudder, there was some effect but still didn't always get it centered. maybe there was another reason for that.

However my Elevator did not respect my trims at all and in the end I had to alter the control horns physically.

But again I would like to ask the question, perhaps I misunderstand. But if I was setting the trims automatically in flight in manual mode on the transmitter, then save those when switching to FBWA, wont those trims on the pixhawk get added to the trims on my transmitter and end of with double the values. it would be ok if I quickly reset transmitter trims, but then there'd be a problem if I needed to switch back to manual, no?

magicrub, 2016-02-23:
>> if I was setting the trims automatically in flight in manual mode on the
transmitter, then save those when switching to FBWA, wont those trims on
the pixhawk get added to the trims on my transmitter
I'm not a FBWx user but I think this is wrong. You want your transmitter to
be all zero'd out when using FBW.

WickedShell, 2016-02-24:
Trims are just a notion of what neutral is. When you change trim on you're Tx all you change is the resultant output signal from the Tx. The aircraft has no idea if you moved the controls or if you changed trim. So what that value is really doing is being used to translate you're stick positions at percentage inputs for FBWx (or stick mixing in auto modes). Manual control will simply be outputting what you put in. The RC trim is set when you do a radio cal and you should never need to bump it outside of that.

You should trim the model to be correct in manual on the Tx, then do a RC calibration to match the value, this will enable good reading of the sticks on input for any of the FBW etc modes. And you won't necessarily see changing a trim cause a surface to move its position (IE you could have been commanding a descent before that/more airspeed) and the small change of commanded value still requires a max deflection to achieve the desired rate.

I'm closing this as this is a perception problem/misunderstanding of what trim actually does.

closed 2016-02-24 by WickedShell

iskess, 2016-02-24:
@WickedShell, I always used the TRIM_AUTO parameter to grab the trim values. Is it safe to say that doing an RC Calibration after manually trimming does exactly the same thing that TRIM_AUTO does? Of course you can't do an RC Calibration while flying.

WickedShell, 2016-02-24:
IIRC TRIM_AUTO resets the trim value to be wherever the Tx is set to when it powers on. It does this every time it power cycles, in general I prefer to do one good calibration and leave it. Otherwise I have to be aware of where the Tx sticks are on power cycle.

iskess, 2016-02-24:
TRIM_AUTO takes the trim that you set while flying around in manual, and writes those values into the RCx_TRIM parameters when you switch to FBWA. It's useful for trimming while flying, but dangerous if your fingers are on the sticks when switching modes.

skystarUK, 2016-02-24:
WickedShell.
This is not a problem with perception, the problem is NOT resolved as far as I'm aware, although I haven't tried the latest beta.

The original post of this issue was of Trims not working.
To quote:
"I am trying to manually change the trims for the aileron and elevator servos on a plane. When I type a new value and click "Write Parameters" there is no change in the servo trim and when I look at the status menu the RC Output is still the previous value even though the RC Trim value has changed."

I have exactly the same issue, which is why someone reopened the issue. My settings show that RC2, RC3 trims are set in MissionPlanner settings, however when watching the Outputs on the servos in MissionPlanner those settings appear to be totally ignored.
RC5_Trim, for my Flaps does work...
I've tried many different values in the RC2_Trim just to test if I could get something to happen but, no.

I in the end had to alter my servo linkages to get it set right, but I am concerned as I am adding more weight with FPV gear I would have liked to be able to adjust trims on the fly with radio or missionplanner.

I only mentioned the problem of trims when switching between modes as a side issue I found when trying to figure out the original Trim ignoring problem.

skystarUK, 2016-02-24:
iskess:
Thanks.  yes thats how I thought it supposed to work.
http://plane.ardupilot.com/wiki/arduplane-parameters/#automatic_trim_adjustment_arduplanetrim_auto

But if I trim'ed out my Bix in manual mode to get it spot on. Then switched to FBWA with Auto_Trim on.
Should I then need to reset the Trims on my tx?  I assume so but , then its messed up if I switch back to manual..?  Is this not a problem for you?

WickedShell, 2016-02-24:
@skystarUK Post a log while changing them that shows the issue, but it still looks like its working right to me. Manual mode it wont have any effect, in FBW it should have an effect but there is also a lot of other things happening in there that could mask it. IE the deadzone could be to large, the surface could already be deflected. I can't reproduce it on any plane I have, and it behaves as I expect it to.

skystarUK, 2016-02-24:
ok thanks for the help. I will try to find more data.

This is not in air testing though, I am still setting up on bench, and hasn't been good enough weather this year so far where I am to fly something new.

On the bench all I am doing is sitting the Plane level, seeing that the Elevator is deflected down too much, so I attempt to alter trim to adjust it. Nothing changes in the servo outputs even though the Trim setting appears to be saved.
I try various power off restarts of all equipment, sometimes I think something has worked, but in the end no..  I would be more like to blame myself if none worked, but since Flaps Trim was set correctly, I obvious did that right..

iskess, 2016-02-24:
@skystarUK,
After setting the trim on the RC and writing those values to the RCx_TRIM Params, then leave the trims alone on the RC. Just remember to turn off the TRIM_AUTO. 
I usually reset the subtrim on the RC so the trims appear neutral, although they are in fact not. 
Does this answer your question?

skystarUK, 2016-02-27:
Well I'm not sure tbh! sry
When you reset the subtrims then you switch back to manual mode which ignores trims set on pixhawk, are the trims not out?

WickedShell, 2016-02-27:
You want the trims on you're RC Tx to match the trims on the autopilot

skystarUK, 2016-02-28:
when in manual mode, but not in pixhawk modes?
I previously didn't like to set specific mode settings on the Tx incase the pixhawk switches mode automatically due to failsafe making it inconsistent with Tx mode setting.
However I guess pixhawk never auto switches back to manual? so it would be ok in that case

WickedShell, 2016-02-28:
I don't see the concern, the trims should just always match? If they don't
match then if stick mixing is on you can have a slow constant input that
you weren't intending.

But yes the plane doesn't just switch to manual ever (unless told to by the
GCS)

skystarUK, 2016-02-29:
Manual mode:
"All RC inputs are passed through to the outputs."
Meaning altering trims would directly alter control surfaces.

FBWA mode:
"In this mode Plane will hold the roll and pitch specified by the control sticks."
In this case the trim would alter the angle of the pitch, so if the trim was not centered the plane would try to angle its self up, maybe a few degrees.?    and would presumably continue altering control surfaces until it achieved that.

So Trim means a different thing in different modes.

As you said before, if you calibrated with pixhawk with the trims set for manual mode that could work, but that would mean a new calibration evey time we needed to alter trim?

WickedShell, 2016-02-29:
As you quoted perfectly in manual the input goes straight to the output. It
doesn't matter what the RC trim on the autopilot is set to.

Yes in general I redo the cal after any changes to trim, but usually not
many of these are required. If you don't fly in a FBW mode or leave stick
mixing off you won't notice the difference either way for a bit, there is
also a bit of a deadzone (RCx_DZ) that is used to allow a small amount of
movement before taking it as an input. It's primarily meant to protect
against jitter on the input but you could raise it a bit to give a couple
of clicks of up down trim without any effect on FBW/stick mixing.

skystarUK, 2016-02-29:
Thanks, very much for your advice. I guess the reason I was so concerned initially is that I'd had to alter the trim a significant amount for elevator on my bix3, and I couldn't alter the servo horn side of it as the servo was glued inside.

I was concerned about switching mode with so much trim and what affect it might have in the different modes.  Which is why I was keen to get the trim settings right in pixhawk.

I've now altered the control surface side (something I didn't do before as I was worried about limiting the range of motion.

I still think I have the same problem was originally mentioned by thread starter. Hopefully you wont mind me querying the issue again when I've got some more data/evidence.

Autonomost, 2016-06-13:
What was the solution? My RCn_TRIM will only trim one elevator (even though both have a separate port and I'm selected different RC numbers) and wont trim my rudder at all...

magicrub, 2016-06-13:
@Autonomost this bug was reported on ArduPlane v3.3.0, is that the version you're having this problem? That is quite old, try the latest released v3.6.0.

Autonomost, 2016-06-13:
Nope it's happening to me on 3.6.0 :/ this was the only thread I found online. I posted something on the ardu forums but they're quiet these days

magicrub, 2016-06-13:
which forum?

Autonomost, 2016-06-13:
http://discuss.ardupilot.org/t/3-6-not-trimming-both-control-surfaces-in-separate-ports/9685/1 

This one.

skystarUK, 2016-11-02:
Sorry I didn't notice your post Anon.
No I didn't solve the problem of setting the trims on Pixhawk manually by altering the settings in Mission Planner.
I gave up on this forum after rather insultingly being told I didn't know what Trim does... Despite the person knowing nothing of my previous RC experience.

The procedure is to setup the trims and throws on the Radio Tx then fly manual to check they're right.
Then do radio calibration in Pixhawk with those settings on the radio. (I had the wrong idea here previously as in Arducopter you setup radio calibration with zero'd trims.)
When calibrating the radio the values for max throws and centre trims get setup on Pixhawk for you.

I was skeptical that this would work properly after having so much trouble trying to setup the trims directly using Mission Planner parameters. But it appeared to work and I've flown it for the first time in FBWA a couple of days ago, successfully.  (I had flown it manually before that)