ArduPilot/ardupilot #2290: Not able to arm with RC3_DZ=0

tuloski, 2015-05-13:
AC 3.2.1
If I put the deadzone on channel 3 (used on throttle) to 0 the quadcopter won't arm.
It could be as small as possible (tried up to 5) but not 0.

R-Lefebvre, 2015-05-13:
Probably not unexpected.  There is always a bit of jitter on these signals of about 1-2 uS, which would mean that while trying to arm, the throttle is in fact, not zero, which it needs to be to arm.  It could even be a physical issue with your joystick, where the PWM transmitted with the stick full down and center, is not the same as the PWM with the stick full down and pushed to the right.

I suspect we don't have an issue here.  To be sure, I'd like some data.  If you go to the radio calibration screen, and hold the stick full down and center, what is the number showing?  Do you see any movement?  It would be normal to see it going up and down a little, such as 1080-1082.  Now, push the stick over full right (be careful not to really arm it with props attached while doing this test!) and down, what value do you see?

tuloski, 2015-05-13:
I see 994-994...same as going full right.
But why should the deadzone affect this? Isn't the deadzone around 1500 us?

R-Lefebvre, 2015-05-13:
No, I think the dead zone on throttle is all at the bottom.  I'd have to
check, but pretty sure.

tuloski, 2015-05-13:
According to the parameter description: "dead zone around trim."
And trim should be the mid point (close to 1500us). So this shouldn't affect the arm process where RC3 is close to 1000us.

rmackay9, 2015-05-14:
Throttle deadzone is actually at the bottom so the issue is perhaps more of a documentation issue because we don't have different parameter descriptions for each rc-channel.  I've slightly modified the parameter description and this should be picked up with the next mission planner update.
https://github.com/diydrones/ardupilot/commit/1b13250fd84ed9197f9a4f1a8a0cd1885839eec1

closed 2015-05-14 by rmackay9