ArduPilot/ardupilot #2298: Documentation for pre-loading SRTM tiles

gmaclennan, 2015-05-15:
We've been operating drones in remote areas with no internet, and often are planning missions in the field with no internet. We need to be able to pre-load SRTM tiles for a large area prior to travel, or to be able to send down a USB drive with the tiles to be transferred to the APM sd card.

I hear this can be done, but would love some documentation about how to do it.

Naterater, 2019-01-08:
Wouldn't this be a function of a GCS? I don't think ardupilot itself is responsible for finding the SRTM tiles. It could probably be done on an SD card, but then this is probably a wiki or GCS issue.

rmackay9, 2019-01-09:
ArduPilot prefetches the tiles required for the mission.  I agree this is more of a documentation issue so perhaps a new issue should be raised in the wiki.  For copter it might go on the [terrain following in missions wiki page](http://ardupilot.org/copter/docs/terrain-following.html).

auturgy, 2020-05-06:
This is documented here: https://ardupilot.org/plane/docs/common-terrain-following.html#common-terrain-following

closed 2020-05-06 by auturgy

magicrub, 2020-05-06:
This is not documented. The question/issue is about how to preload onto the sdCard. I've done this before, you can put the whole Earth on there. Needs documentation.

auturgy, 2020-05-06:
Ok. Let’s raise an issue in the wiki then, rather than here.