ArduPilot/ardupilot #2300: Disarming in Drift Mode leads to flip the Copter on the ground

Gregmaan, 2015-05-16:
I know its silly to land in drift, but when done, disarming via yaw should not lead to flip the copter on the ground.

kmd1984, 2015-07-21:
I wanted my friends to experience drift mode, so I handed him the controller and then pressed the A button, which enabled drift mode. He flew around a little bit. When it then started to rain, I told him to hold still so I can hit fly again, in order for me to take over. FLY would not work though. It could have been because it was overcast and I didn't have enough SATS in sight, who knows?! Anyhow, I just told him to hand me the controller. I then landed. However, keeping the throttle all the way down did not turn the motors off? Instead it flipped?

Here is the log file. have a look.

http://www.rcgroups.com/forums/showpost.php?p=32199826&postcount=8850

rmackay9, 2015-07-21:
@kmd1984, thanks for the log although Solo only produces tlogs which are pretty useless for diagnosing the issues - that logging issue will be resolved in a future release though i hear.

By the way, Solo uses a modified version of AC3.3.  They have plans to get back onto master but I can't make any promises that if we resolve the issue in master when it will be fixed on the Solo branch so feel free to also raise the issue through the Solo app.

I'll do a test of landing in drift mode with ardupilot-master on AC3.3 to see if it does flip though.

kmd1984, 2015-07-21:
Randy,

Thanks for the info and that you are going to try it yourself... I uploaded the bin of that flip to mediafire... Would the bin be of more help?

http://www.mediafire.com/download/h3zc8gtsntzs7c9/Log_of_Flip.rar

Naterater, 2019-01-07:
Has this been solved? This issue is a few years old.

rmackay9, 2019-01-08:
Yes, I think vehicles can land and disarm in drift mode now.

closed 2019-01-08 by rmackay9