ArduPilot/ardupilot #2307: Copter landing gear servo output parameters not documented in the source

hamishwillee, 2015-05-18:
The auto-generated copter parameters for the servo AUX-to-purpose mapping do not include options 28 and 29 (landing gear).

For example, see [RC10_Function](http://copter.ardupilot.com/wiki/configuration/arducopter-parameters/#servo_out_function_rc10_function) - this stops at number 27. However we know that option 29 is the landing gear, and presumably option 28 is something too. The other RCn_Functions are also missing this information.

R-Lefebvre, 2015-05-18:
This is simply a MP release timing issue.  It will be rectified when AC3.3 goes out stable.

hamishwillee, 2015-05-18:
Thanks. Closing then.

rmackay9, 2015-05-19:
I don't think this will be fixed with a MP release.  Hamish is talking about the automatically generated wiki pages so very separate from the MP.

hamishwillee, 2015-05-19:
Correct. I didn't read Robert's message properly :-0 . Normally updates to these parameters go out as soon as they are in the main codeline so I would expect them there now.

R-Lefebvre, 2015-05-19:
Ah, sorry about that.

rmackay9, 2015-05-27:
Ok, found the issue and fixed.
https://github.com/diydrones/ardupilot/commit/b121d83beb28fd0fcf4ff9dc0e10186d9e484d55

closed 2015-05-27 by rmackay9

hamishwillee, 2015-05-27:
Thanks - I can see the parameters are no on the wiki!