ArduPilot/ardupilot #2312: RATE_YAW_FILT_HZ reset to 5Hz on rebooting

R-Lefebvre, 2015-05-20:
This parameter defaults to 5Hz which is much too low for helicopters (should be closer to 20Hz), and even if set to 20Hz, it seems to keep getting reset to 5Hz on every boot.  This is a real problem for helicopters, but I'm surprised if this is the correct action for multirotors too?  I thought this was supposed to be tunable?

This is a must fix for AC3.3

R-Lefebvre, 2015-05-20:
This is extremely bizarre, and I can't begin to explain it, but I've tried this on 3 different boards.  If you change the parameter to 20, write it, refresh params to make sure it went in, and then reboot, it keeps getting reset to 5Hz.  No matter how many times you do this.

However, if you set it to 30Hz, write it, reboot, then it sticks and does not reset to 5Hz.  Then, if you set it to 20Hz, write it, reboot, then the 20 sticks!

R-Lefebvre, 2015-05-20:
Jon called me about this issue and we looked at it.  Seems the problem is that:

#define AC_PID_FILT_HZ_DEFAULT  20.0f   // default input filter frequency

AP_GROUPINFO("FILT_HZ", 6, AC_PID, _filt_hz, AC_PID_FILT_HZ_DEFAULT),

So when I set it to 20, it's not actually storing it in FRAM because I've just set it *too* the default value.  Then on reboot, since it hasn't actually been saved in FRAM, it reverts back to:

 # define RATE_YAW_FILT_HZ          5.0f

Now, we can paper over this, by #define the default to tradheli to 20.  Just not sure if that's the best solution.

jschall, 2015-05-20:
Another solution is to change AC_PID_FILT_HZ_DEFAULT to -1 or some other bizarre value, which would keep the logic in AP_Param::save from refusing to write 20 to FRAM if there is no value yet stored in FRAM.

jschall, 2015-05-20:
Note that this might make sense for the PID values as well. I can imagine a user wanting to set one of them from the ctor default to 0 (the static memory default) and it fails and then they crash. Alternatively, we could move away from constructor defaults and instead use something similar to the default override file for these parameters.

R-Lefebvre, 2015-05-22:
Fixed in this PR:

https://github.com/diydrones/ardupilot/pull/2318

closed 2015-05-22 by R-Lefebvre