ArduPilot/ardupilot #2333: Deactivating tone_alarm (buzzer) in rcS causes startup freeze

deb0ch, 2015-05-25:
Hi guys,

I use and develop on an Iris+ quadcopter for a few month now, therefore I keep flashing and rebooting it.

The thing is, my colleagues and friends were so close to hang me or do many other kinds of nasty things the last time that they heard the startup tune, that I decided to finally do something and deactivate it. But for now without cutting the wires.

So I went in the mk/PX4/ROMFS/rc* files and just commented all the calls to 'tone_alarm start'.

Doing that just made the hole thing crash: at startup, instead of the blinking and color-changing LED it is just fixed white. I also have nothing on the serial port, nothing nowhere: the hole system just does not work anymore.

Does anybody know the reason for that ?

magicrub, 2015-05-25:
Sounds like you commented out some critical flags. I suggest only
commenting the code that triggers the sound and leave the flags alone.

Easy solution is to simply unplug the speaker which is what everyone else
does who wants it off.

deb0ch, 2015-05-25:
No, I really just commented each one of the 3 occurrences of 'tone_alarm start' in these files, and only these lines.
The problem seems to be with the exit status of the program. I hacked a little bit into the sources of tone_alarm, in src/drivers/stm32/tone_alarm/tone_alarm.cpp and modified in the play_tune function the exit code given to the exit() function. With non zero values it fails, with a zero value it succeeds. Though, replacing 'tone_alarm start' with the shell command 'true' to get a zero exit status did not work either.
I find the other beeps helpful, don't get me wrong, I just can't stand the startup tune anymore, and neither can the people around me.

magicrub, 2015-06-05:
I'm closing this because your code change is not an "issue" with ArduPilot. Please create a new issue with the real problem you're trying to address: "I just can't stand the startup tune anymore, and neither can the people around me."

closed 2015-06-05 by magicrub

rmackay9, 2015-06-06:
There's already an item on the to-do to add a MUTE.  Many people would love that addition.