ArduPilot/ardupilot #2334: Remove 4 second pause before landing command is activated during EKF failsafe

rrr6399, 2015-05-26:
I was recently flying a copter with firmware 3.2.1 using a GCS. The copter was flying in AUTO mode and was in unlimited loiter and a target heading of 90 degrees. For some reason that I'm still trying to figure out, it took off across the field. I immediately switched it to LAND mode, but according to the logs, that caused an EKF failsafe and it appears form the code below that the land command was delayed 4 seconds. I wasn't aware that there was a delay and the copter was heading for trees so I killed the motors resulting in a hard crash. 
https://github.com/diydrones/ardupilot/blob/master/ArduCopter/ekf_check.pde#L132
https://github.com/diydrones/ardupilot/blob/master/ArduCopter/control_land.pde#L203

I don't know the history behind this 4 second  delay, but it is an eternity during an emergency situation. In retrospect, I may have switched it to stabilize, but if there the 4 second delay could be removed, I believe it would allow more robust operations from a GCS during  emergency situations.

rrr6399, 2015-05-26:
Here is the altitude and EKF error history:
[image]

rrr6399, 2015-05-26:
Here is the yaw target and yaw measurment. Not sure why it is ramping down like that. The yaw target was set to a constant 90 degrees.
[image]

rmackay9, 2015-05-27:
The crash was likely caused by a heading error (bad compass, bad gyro calibration, etc).  The bad heading would have been detected by DCM or EKF and the EKF Check would have triggered the Land.

The 4sec delay is not a technical question so much as a UX question.  The delay was added after multiple people complained that they had no time to react to a failsafe induced land before the vehicle started descending (into the water, into the trees, etc).  I think it's unlikely we can get rid of the delay without a consensus that it's addition has done more harm than good.

rmackay9, 2015-05-27:
I think the way to get that consensus is to email drones-discuss@googlegroups.com and get a few people to agree.
Sorry to be annoying but I'm going to close this issue for now because there's no way to act on this request without that consensus.

rrr6399, 2015-05-27:
I actually commanded it to Land and it apparently delayed my request due to this logic. Its almost like the EKF failsafe logic was activated by changing the mode to Land.

Its possible that it hit this failsafe right before I commanded it to land, but the odds seem low.

I will take a look at the log to see if the land command got through at the exact time as the EKF failsafe was signaled.

rrr6399, 2015-05-29:
Here are the results from the auto-analysis capability in Mission Planner:
Log File C:/Users/Rob/AppData/Local/Temp/tmp9892.tmp.log
Size (kb) 527.400390625
No of lines 7614
Duration 0:00:39
Vehicletype ArduCopter
Firmware Version V3.2.1
Firmware Hash 36b405fb
Hardware Type 
Free Mem 0
Skipped Lines 0

Test: Autotune = UNKNOWN - No ATUN log data
Test: Balance/Twist = GOOD - 
Test: Brownout = FAIL - Truncated Log? Ends while armed at altitude 13.47m
Test: Compass = GOOD - mag_field interference within limits (24.80%)

Test: Dupe Log Data = GOOD - 
Test: Empty = GOOD - 
Test: Event/Failsafe = GOOD - 
Test: GPS = GOOD - 
Test: IMU Mismatch = NA - 
Test: Parameters = GOOD - 
Test: PM = GOOD - 
Test: Pitch/Roll = GOOD - 
Test: Thrust = GOOD - 
Test: VCC = GOOD -

rrr6399, 2015-05-29:
It looks like the LAND mode change I issued stimulated the EKF FAILSAFE:
[image]

I think there is a bug in the logic and prevented my LAND command from taking effect until too late.

rrr6399, 2015-05-29:
Randy, would you reopen this. I believe there is an actual bug in the firmware related to issuing the LAND command and it doesn't require a policy decision.

magicrub, 2015-06-05:
@rmackay9 Please re-evaluate and re-close if necessary.
@rrr6399 Per Randy's suggestion, please open up a drones discuss email.

rrr6399, 2015-06-05:
It appears from the logs of two failed fly away flights with two separate copters that when the pilot initiated the LAND command via the GCS that the LAND command triggered the EKF failsafe logic not vice-a-versa. This failsafe logic has a 4 second delay logic that I suspect is causing the delay in the LAND command. Since it appeared that the copters were out of control given that the change to LAND mode was being ignored by the firmware, the both copters were disarmed in flight leading to hard crashes.

rrr6399, 2015-06-05:
Here's altitude and history of the copter that crashed, which also didn't respond to the GCS commanded LAND command. Notice the EKF errors as well when the Land mode was initiated.
[image]

Here's the full flight:
[image]

rmackay9, 2015-06-06:
Is there a log file somewhere?

rrr6399, 2015-06-07:
@rmackay9, I'll get you log files. Are you looking for anything particular in the results?

rmackay9, 2015-06-09:
Here's the competing request asking to go in the other direction (i.e. increase the delay): https://github.com/diydrones/ardupilot/issues/1905

magicrub, 2015-06-09:
Perhaps a time param will make everyone happy?

rmackay9, 2015-06-09:
To some extent I think the basic problem is that the pilot doesn't know the status of the vehicle and no matter how long the delay is or isn't it will not properly convey the situation.  I think increasing pilot awareness is best done through a ground station or transmitter that can yell out alerts to the pilot especially in these critical cases where control is passed between pilot and autopilot.

rrr6399, 2015-06-09:
I think one thing that is being missed in this case is that I commanded the copter to Land and it ignored it for 4 seconds. That is different from the failsafe scenario. I believe that for some reason, the EKF logic was activated by the GCS's Land request.

peterbarker, 2021-07-30:
We now have `set_mode_land_with_pause(ModeReason::EKF_FAILSAFE)` to cover this.

closed 2021-07-30 by peterbarker