ArduPilot/ardupilot #2342: Reduce final stage of LAND as rate of descent is still to high

dcwalmsley, 2015-05-28:
I'm writing this as I have observed that every time my copters come down from a temporary hover (approx. 10ft) they descend too quickly and have on a couple occasions tipped over on uneven ground.  
I have LiDAR units on both vehicles and would like the descent speed reduced at the last few feet based on LiDAR sensing to help provide a smooth and easy landing.  Not sure LiDAR is needed for this idea, but as I see it now I'm having issues with descent rates on final landing for over a year or more.  Anyway to adjust this in code or mission planner?

R-Lefebvre, 2015-06-01:
Have you tried adjusting the parameter LAND_SPEED ?

dcwalmsley, 2015-06-02:
I'm an idiot but I ran acrossed it last weekend. Please disregard.

rmackay9, 2015-06-02:
i'm sure you're not.
Closing this issue!

closed 2015-06-02 by rmackay9