ArduPilot/ardupilot #2346: Yaw Twitch After Takeoff

R-Lefebvre, 2015-05-29:
Yaw twitches when taking off, twice, first at 1.5m and then at 5m.  This is because of the EKF dealing with the possibility of ground magnetization.  Very noticeable with tradhelis.

Paul has the fix for this already, the yaw reset is slowly integrated over a short period of time, just needs to be merged in.

robustini, 2015-05-29:
Same issue here, confirmed!

R-Lefebvre, 2015-05-29:
Yeah, it's a well known issue that's already got a fix.  This is just a reminder to pull the fix in.

As of now, the yaw will still move a little bit, but it won't be so aggressive.

What's happening is the yaw heading resets after leaving the ground, which is meant to fix problems that occur when people take off from a surface that is magnetized.

andrea25, 2015-06-02:
Confirmed also by me. Same issues here. Very noticiable also with Hexa.

rmackay9, 2015-07-07:
this is fixed with AC3.3-rc7

closed 2015-07-07 by rmackay9