ArduPilot/ardupilot #2347: MicroSD full, stop writing logs.

robustini, 2015-05-29:
After about 90 logs my MicroSD 4 Kg is full (64k free), and Pixhawk has stopped writing any log, don't delete the old ones.

R-Lefebvre, 2015-05-29:
Are you requesting that it should start over-writing the old logs?

magicrub, 2015-05-29:
That's a heavy sd card!

robustini, 2015-05-29:
@R-Lefebvre: yes...
@magicrub: LOL! :P

WickedShell, 2015-05-30:
I mostly have deleting old log files if there is insufficient space at
startup working (tries to delete the oldest logs until at least 10% of the
SD card is free). However I'm experiencing a somewhat subtle bug with the
filesystem (solely related to my changes) and haven't gotten to finish
debugging it yet.

This doesn't help with filling logs in flight, but should help make it less
likely as long as you don't log to much or fly for a long time. (Assuming
the 4gb card from 3dr)

rmackay9, 2015-05-30:
@WickedShell, yeah, a check at startup to ensure x% is free or an absolute size would be best I think.

robustini, 2015-05-30:
A log can also be over 100 Mb, I think an "absolute free space" is necessary, and in that case when the code writing a log must delete the oldest to keep that space always free, for example 20%.

WickedShell, 2015-05-30:
In the end after having talked it over with Tridge in the past it seems that a percentage probably works best. IE requiring 10% free at boot. With a stock 4GB card from 3DR this would provide just over 400MB's for logging in the flight this should be more then sufficient for most users, and if you are using a smaller card then presumably you need less log space anyways, and in reverse on a larger card this will grant you more logging space as guaranteed. Some of my flights have generated 700+MB logs which is well outside of a typical users log size, and typical log flights easily produce 200MB logs for me. However most copters don't fly nearly as long so they produce smaller logs.

@robustini the logic used was to keep deleting the oldest log left until 10% is free (even if that means deleting all the logs due to other user data on the sd card (terrain data, pictures, misc stuff on the sd card (mission planner is always a nice one to have there)). And it's only looking at erasing logs during the boot process, if it runs out of log space whilist in flight then thats the end of logging for the flight.

peterbarker, 2015-08-08:
PR to remove files until 10% free space: https://github.com/diydrones/ardupilot/pull/2678

rmackay9, 2015-09-09:
Peter Barker's fix above is in master and resolve this issue.  It's not in Copter3.3.

closed 2015-09-09 by rmackay9