ArduPilot/ardupilot #2352: DataFlash: Downloading logs over MAVLink lists no logs if the SD Card is full

WickedShell, 2015-05-30:
If the SD card was full at boot time, then asking the autopilot to download or clear logs over MAVLink will show no logs on the autopilot.

Tested against Pixhawk running APM:Plane 3.3.0 with the stock 3DR 4GB SDCard and Mission Planner 1.3.28

peterbarker, 2015-07-01:
I believe the problem here is that the highest-numbered-log is 0-bytes long, rather than that the log card is full.  One could cause the other, of course.

This PR might fix the problem: https://github.com/diydrones/ardupilot/pull/2485

Can you check that, please?

peterbarker, 2015-11-25:
@WickedShell This works for me now.

closed 2015-11-25 by peterbarker

WickedShell, 2015-11-25:
Oh jeez, I never saw the comment that you had a PR that could fix the problem... Sorry! (will test shortly, but if it's working for you I suspect it will be good here)