ArduPilot/ardupilot #2353: Feature request: Add parameter to ArduCopter to control when disarm command is accepted/rejected

apogeecmb, 2015-05-30:
During some recent flights, we noticed that in some modes (at least RTL and Auto) a command to disarm while in flight is now rejected, both from Mission Planner and when attempting to disarm using the RC.  I assume this feature was put in to prevent people from accidentally disarming their vehicles in flight and crashing them. 

However in our particular application, we use disarming as a flight termination system if a flight test goes wrong and a vehicle becomes uncontrolled and a danger to persons or other objects. We did find a workaround for this by first switching to Stabilize and them sending a disarm command, but having to take this extra step is not ideal for our application.

I would like to request adding a parameter to enable/disable this disarm lockout feature. I understand why it is in place, but I also think it would be nice to be able to turn it off. Thanks!

WickedShell, 2015-05-31:
I think this might be a good call against all vehicles, but since I haven't ever looked into how the other vehicles handle arm/disarming and the issue is against copter I've only tagged it as copter for now.

rmackay9, 2015-05-31:
This is fixed in AC3.3.  If the ground station sets parameter 2 to 21196 (a "magic" number that is incredibly unlikely to be set accidentally) it will disarm without performing the landing check.
We need to add this to the MAVLink spec so that it's offcial and people have a better chance of finding it.

closed 2015-05-31 by rmackay9

apogeecmb, 2015-06-01:
Randy,

Thanks for the quick feedback on this.  This is really a question for the GCS developers, but would it be your expectation that they would want to use this new addition to the disarm command that skips the landing check or continue to use the current command?  I'm wondering if they'd keep it as is to prevent things like accidental disarms (although Mission Planner at least already pops up a confirmation window any time you click arm/disarm).  Just trying to think if maybe this still might be something that should be a configuration parameter you can disable.