ArduPilot/ardupilot #2356: error: 'AP_AHRS_NavEKF' does not name a type

nscnm87, 2015-06-01:
Hi,
when I try to do "make flymaple" in ArduCopter directory the terminal say:
```
     %% control_rtl.o
     In file included from /home/rey/ardupilot/ArduCopter/control_rtl.cpp:3:0:
     /home/rey/ardupilot/ArduCopter/Copter.h:183:5: error: 'AP_AHRS_NavEKF' does not name a type
     In file included from /home/rey/ardupilot/ArduCopter/control_rtl.cpp:3:0:
    /home/rey/ardupilot/ArduCopter/Copter.h:410:5: error: 'AP_InertialNav_NavEKF' does not name a    type
```

I haven't any kind of problem with ArduPlane or APMRover2.

I'm on Ubuntu 15.04 and I've ran the script "install-prereqs-ubuntu.sh"

nscnm87, 2015-06-30:
There is a way to compile with Windows?

WickedShell, 2017-02-10:
Flymaple support is deprecated, and no longer on master. (However we appear to emit a wrong warning with make)

closed 2017-02-10 by WickedShell