ArduPilot/ardupilot #2357: Bug: Land command initiated by pilot is ignored due to EKF failsafe logic

rrr6399, 2015-06-01:
Since it doesn't appear that I can reopen this bug myself:
https://github.com/diydrones/ardupilot/issues/2334 ,
 I created this bug.

It appears from the logs of two failed fly away flights with two separate copters that when the pilot initiated the LAND command via the GCS that the LAND command triggered the EKF failsafe logic not vice-a-versa. This failsafe logic has a 4 second delay logic that I suspect it is causing the delay in the LAND command. Since it appeared that the copter was out of control since the change to LAND mode was being ignored, the both copters were disarmed in flight leading to a hard crash.

magicrub, 2015-06-05:
@rmackay9 , please re-evaluate https://github.com/diydrones/ardupilot/issues/2334
@rrr6399 , please do not intentionally open up duplicate issues. If there is no response to comments on an issue then bring it to the attention of the gitter chat channel or drones-discuss email.

closed 2015-06-05 by magicrub

rrr6399, 2015-06-05:
The issues are being closed too quickly. It takes a lot of time to generate a decent bug report and its frustrating to see the issue immediately closed without any further discussion. There is no way for me to reopen the request either.