ArduPilot/ardupilot #2359: Add Parachute to plane

CraigElder, 2015-06-03:
Add
https://github.com/diydrones/ardupilot/blob/master/libraries/AP_Parachute/AP_Parachute.cpp
enable deployment from a mission command
stop the motor as part of the deployment sequence

magicrub, 2015-06-03:
What is the current deployment method in Copter?

alexchernoff, 2015-06-04:
Suggestion on chute opening on Plane:  if a DO_SET_SERVO command exists after DO_LAND_START, assume it is a chute servo. Once the aircraft enters the Flare stage and altitude, it cuts off throttle and executes the servo command. 

This way:
1: aircraft is very close to home and on approach
2: it's at a flare (lower) altitude than RTL altitude.

rmackay9, 2015-06-04:
In case it helps there's a MAV_CMD_DO_PARACHUTE that can be used to trigger the parachute as part of a mission (in copter at least).  It should also be possible to send it wrapped in a COMMAND_LONG so that it can be triggered immediately from the ground station.

In addition in copter we use ch7/ch8 aux switches to allow the pilot to trigger the release manually.

naktinis, 2015-07-14:
Since #2553 was merged as duplicate, I wanted to stress that it was particularly about automatic deployment (what Randy referred to as "crash checker") when some criteria are met (not about deployment from a mission command). Similar to what's implemented for the copter.

rmackay9, 2015-07-14:
I've updated the title a bit to hopefully make that clear.

tridge, 2015-10-27:
initial version of this is now in master

closed 2015-10-27 by tridge

naktinis, 2015-11-02:
Since this is now closed, I think #2553 should be reopened. The code that is now in master, as I understand, does not automatically detect a fault and deploy a parachute (which Copter is capable of and which #2553 is about).