ArduPilot/ardupilot #2368: Plane: RC7_OUT stops working mid flight using v3.3, OK on v3.2.1

GrahamDyer, 2015-06-04:
On a few flights recently the RC7_OUT (which controls camera pan in our case) went low and refused to respond to RC7_IN. This only happened since updating to APM:Plane v3.3 on two different planes. All wiring is fine and the problem is fixed by a simple reboot but obviously the plane has to return from it's mission, land and be rebooted.

[image]
(Late during this return flight RC7_OUT went high for some reason but still did not respond to inputs.)
I'll include the log file as soon as it's uploaded to Droneshare.

GrahamDyer, 2015-06-04:
https://www.dropbox.com/s/63gae0a2u4xroyk/132.BIN?dl=0

GrahamDyer, 2015-06-07:
Downgraded to v3.2.1 yesterday and had a 90 minute flight with no occurrence of the problem.

gmorph, 2015-06-20:
Are you able to replicate it with 3.3 on the bench?  Can you try a different RC output?  I'm just interested if its something specific to RC7 or if its something to do with the camera control generally.
Thanks, Grant.

gmorph, 2015-06-21:
It was indeed a bug.  Fixed in this pull request.
https://github.com/diydrones/ardupilot/pull/2441
Thanks Graham - REALLY appreciate you bringing this bug to our attention.  Great work.
Thanks, Grant.

closed 2015-06-21 by gmorph