ArduPilot/ardupilot #2374: ArduPlane: Spline for mapping missions

rsesa, 2015-06-05:
I've red in AC 3.2 that spline was already availabe for ArduCopter. I'm trying to perform mapping missions with a Skywalker X5, and even using overshoots of arround 50m, it takes at least 80m to get on the path (see image below). I guess setting spline waypoints at the entry waypoints will largely enhance the mission performance, so when I found out it was not supported on ArduPlane I was quite disapointed, I tryied to write a spline mission in SITL and it just seems to skip spline waypoints.
I think spline is really necessary for the ones like me trying to map, but also I'll be gratefull if anyone with same problems know how to fix or reduce this overshoot by other means.
[image]
Plane came from bottom right corner, already in the air, performs the mission and auto lands.

magicrub, 2015-06-05:
I'm not sure how spline will help in a mission like this. I'm (slowly) working on https://github.com/diydrones/ardupilot/issues/2222 which will make the turns a bit smarter for mapping but not sure if its what you're looking for. Spline is for smoother turns which is nice for videos and such but I'm not seeing much benefit in mapping where you want to be flat and level most of the time instead of always turning

rsesa, 2015-06-07:
Surely this would help getting the track sooner, but if my conception of spline is right it would be much more interesting (when talking about mapping). 
My idea was to set the entry points (WP&, WP8 and WP10) as spline waypoints, so that when the plane reaches de waypoint it would be almost over the track and almost with the right heading for the next leg, which is the one we want to map.
[image]

WickedShell, 2015-06-07:
I would be worried about what happens if your plane can't maintain the
profile of that spline. ie if the spline defines a path that is simply
outside what the plane can handle.

My personal experience has been to simply oversize the polygon as well as
leverage the overshoot numbers. Depending on camera/plane configurations
this can be really clean. I mean in some sense that's really all the spline
waypoint is trying to do here, move out the entry to the next leg so that
it happens outside of the target mapping area.

rsesa, 2015-06-07:
I don't think (from my little experience) that it could be worst that trying to follow the actual flight path (stright lines with 0 radius turns), moreover lane separation can be used to smooth spline turns.

rsesa, 2015-06-09:
How much do you, @WickedShell, oversize your polygon?

WickedShell, 2015-06-09:
Usually at least ~100 meters. It really depends on how your plane is setup, and what altitude you are flying at. For example with the plane and camera we typically use flying at 300 feet with 65% sidelap we need to oversize the polygon by about 125 meteres (or more if there is a strong wind) combined with a 30m overshoot. However flying at 500 feet with the same camera and sidelap we need maybe 20 meters of oversizing.

I should note that on all of this oversizing we could achieve a usable result with less overshoot, but I vastly prefer having a decent amount of extra imagery that can be used to crop down to the desired final result. (Indeed normally our mosiac will extend all the way out to the banks). The only real reason that I tend to see forcing the oversize is typically driven by either how close the flight lines are, or how strong the wind is. Having the plane turn downwind on extremely windy days can cause real problems (or if you don't have a crosswind and are seeing a really large ground speed).

So since our mapping flights are out to three hours in a variety of winds we build our flight plans really defensively, such that no matter what environmental factors happen during the flight as long as we can still fly the mission loaded is fine.

Hopefully that helps? (In some ways I'd be more interested in a series of way points being added that made a teardrop shape rather then splines, or a box). This is much more similar to what manned aviation does, and it works really well for controlling reentry to a desired path.

rsesa, 2015-06-15:
Thaks for your help!

We went flying and tested at 90 m height with 65% sidelap and 50 m overshoot, I'm waiting now for the final mapping to see how many polygon area is lost due to aircraft turning, but we noticed a big improve on turning performance by decreasing the L1 period only by 1 (maybe we sould try decreasing it more).

You idea of a teardrop sounds also realy interesting on what I'm trying to archieve, I don't have enough experience to say wich one is better (yours or mine ;) ), so I'd better trust you!

BjoernKellermann, 2015-08-26:
I would recommend a parameter that defines the diameter of splines a plane/copter can handle based on it's velocity.

With that information read from the flight controller using mavlink, the mission planning software could calculate a grid the aerial vehicle can fellow.

The grid would depend on the arial vehicle's capabilities (diameter for speed) and the intelligence of the mission planner, e.g. For narrow lanes an line interleave path could be generated or the teardrops at every lane end...

This would apply mainly for planes but also for fast moving and heavy copter and might be based internally on total energy model.

Same function also be used to verify a manually greater flight plan. This validation could be programmed into to flight controller to avoid not possible moves and stall prevention. It would not accept a not realistic waypoint or target in guided mode.

Further extension would be altitude validation based in a second parameter for climb rate.

BR

Bjoern

magicrub, 2015-08-26:
>>I would recommend a parameter that defines the diameter of splines a
plane/copter can handle based on it's velocity.

I think a velocity based wp diameter is generally a good idea, spline or
regular.

WickedShell, 2019-01-18:
We still want this as there are valid use cases.