ArduPilot/ardupilot #2376: Copter: does MNT_MODE need to be persistent?

sgofferj, 2015-06-05:
I have been scratching my head today on a filming trip, wondering why I can't suddenly control the gimbal via the TX. I quickly rewired my IRIS for CH6 passthru in the field. At home I figured it out. When setting up the gimbal, I used the testing method described in the wiki and used the "Point camera here function" which set MNT_MODE to 4 (GPS). The parameter is persistent over a reboot and hence the RC-control didn't work.
I was wondering if it made sense to have MNT_MODE default to RC control after reboot.
I also opened issues for MP and AP2 to add a convenient button for MNT_MODE which would also be an alternative.

rmackay9, 2015-06-06:
I guess you're using AC3.2.1?
In AC3.3 MNT_MODE has been replaced with MNT_DEFLT_MODE.  The mount switches back into this mode fairly often.  I think it happens whenever the flight mode is changed.
I'm going to close this for now because I think AC3.3 resolves the problem.  If you could test with AC3.3 and report back especially if it's not ok and we can then re-open.

closed 2015-06-06 by rmackay9

billbonney, 2015-06-06:
OK, great. You are saying MNT_MODE is ignored by AC3.3? This mean I can leave the MNT_MODE instruction in the code for backwards compatibility. 
I'll have to think about how the Camera Config View needs to be modified. Having a way to read the version based on a parameter would be gapped (or feature flag). Thx

rmackay9, 2015-06-06:
Yes, MP's mount setup screen also needs some work and I was planning on creating a screenshot of how I think it should look.  The biggest thing that's missing is we now have a MNT_TYPE parameter which needs to be set to say which mount type you're using (servo, SToRM32, etc) and then the board must be rebooted so i was thinking a little "reboot now" button would be good to place near that drop-down.

avivyam, 2016-04-02:
Hello, new to tihis amazing world,  runing plane pixhawk and tarot2d.
I can control my tarot manualy if i use passthrough on ch9 ( the taraot is on aux1 as the wiki).
When i set everithing as described ( ch9 set as tilt , and ch7 as the input chanel). I can control the gimball with do set roi on the mp. But cant get the mount to respond to the rc. 

I have tried using the set mount to rc control in the actions tab with no sucsess.

 I could not find a way to check mnt_type.

 Will seting the defalt mnt mode to 3 and switching flight modes give me the default mmt mode ? 

  Thanks , 

Aviv