ArduPilot/ardupilot #2377: enhancement: calculated rangefinder altitude from raw reading

srobi, 2015-06-05:
I suggest add an optional function (either as part of RNGFND_FUNCTION or  separately) to re-calculate the rangefinder altitude from the raw reading. Altitude measured by rangefinder assumes horizontal positon of the copter/plane. As Roll/Pitch changes the measurement become non-vertical and causes drop of altitude – rate of drop is increasing with the grade of roll/pith and the altitude (45 degree / 40 m is approx. 11.7m altitude drop that is quite material).

CalcAltitude = Sine(Radian(90-AbsoluteValue(Maximum(pitch, roll))))*RawAltutude
where pitch, roll are the reported angle in degree of the plane/copter
RawAltitude is the raw reading from the rangefinder

magicrub, 2015-06-05:
In plane that calculation is done here
<https://github.com/diydrones/ardupilot/blob/master/ArduPlane/altitude.cpp#L560>.
I assume copter is doing something similar but I agree that it should be
moved to the rangefinder library so it is done more uniformly. Pull
requests welcome.

-TomP

EShamaev, 2015-06-08:
I am working on exact principle rewriting most of all landing code. Will be ready for test in few weeks.

EShamaev, 2015-06-08:
Sorry, wrong thread

rmackay9, 2015-06-09:
Copter has a #define that turns the compensation on/off.  It depends upon the sensor used, so a wide beam sonar should not use compensation while a narrow beam lidar should be.  So we probably need the driver for each sensor to specify whether it needs the compensation or not.

jeffyromney, 2015-06-30:
I would suggest putting this as a parameter because many different sensors use the same analog driver making a driver-based selection impossible. At least for the generic drivers.

magicrub, 2015-06-30:
That's a good point, @jeffyromney. Same goes with PWM input method.

rmackay9, 2015-07-01:
Yes, I agree it needs to be different for different sensors - probably instead of a parameter it should be a method on the range finder backend.

jeffyromney, 2015-07-01:
yes, the calculation itself would be a method on the backend, but we'd need a parameter to specify when to use that method. Hardware specific drivers could ignore the parameter because it is known, but the generic drivers need to be versatile.

Maybe a parameter for the field of view of the sensor would work best. that way with steep bank angles (beyond the set field of view) it would start compensating. setting the field of view to 0 would always compensate(such as for a laser rangefinder). it could use something like this as the compensated roll angle and a similar one for pitch:

comp_roll = (abs(roll) > (FOV / 2)) * (roll - ((FOV/2) * (roll / abs(roll))))
comp_pitch = (abs(pitch) > (FOV / 2)) * (pitch- ((FOV/2) * (pitch/ abs(pitch))))

There might be a simpler way but this is what came to mind to make it as simple and versatile as possible for the end user.

rmackay9, 2015-07-01:
@jeffyromney, yes, that's a good idea.. so you only start tilt correcting after a certain angle.  the sonar would be really wide, the laser much more narrow.  I'd like to avoid a parameter if possible.. I would have thought we could just set the FOV based upon the driver.  So the driver for the lidar-lite would hard-code it to 5deg.  The maxbotix sonar to perhaps 30deg.  It does depend a bit on which sonar is being used but.. not sure it matter that much in practice.

jeffyromney, 2015-07-01:
The driver based method would work for hardware-specific drivers such as the maxbotix and the lidar-lite drivers, but for the generic analog/pwm it won't.  for example, I have a laser rangefinder and a sonar that both work use the same analog driver.  
This is why the following parameters exist:
RNGFND_MAX_CM
RNGFND_MIN_CM
RNGFND_OFFSET
RNGFND_SCALING

with a hardware-specific driver these also aren't needed but they are used by the analog and PWM drivers for versatility. I think this field of view parameter would be in the same category here.

srobi, 2015-07-01:
For exact measurement and terrain following function people may use rangefinder on a gimbal instead of fix vertical mount (to measure the altitude right below the copter not side/behind) - as I do my Lidar. So tilt correction should be optional I would suggest.

jeffyromney, 2015-07-01:
setting to -1 could disable it. also setting a 360 degree fov would effectively disable it.

nicksargeant, 2016-06-26:
I have reviewed Plane (altitude.cpp) and Copter (sensors.cpp) and it appears that pitch and roll compensation for rangefinder altitude measurements is being conducted. On copter, it appears that compensation can be disabled (RANGEFINDER_TILT_CORRECTION) in config.h. Can I please confirm from the devs that these assumptions are the correct? In the case of plane, Is the distance in the dataflash logs (RFND.Dist1), the corrected altitude? Thanks!

magicrub, 2016-06-26:
the value in RFND.Dist1 is the raw value form the rangefinder, it is not
tilt compensated. For Plane, the compensated and filtered value that is
actually used can be found in SONR.Dist

rmackay9, 2016-06-27:
@nicksargeant, you're right for the copter part at least (I'm unsure about plane).  Recently while adding terrain following support during missions into copter. I turned on the tilt correction by default.  We should probably have a parameter to allow setting it per sensor.

nicksargeant, 2016-06-28:
Thank you both for the responses! Tom, SONR.Dist only appears to have come into effect since Plane 3.6. Can you please confirm? Do you mind having a look at the following logs (zipped) of  Plane 3.5 and Plane 3.6 with the lightware rangefinder? RFND.Dist1 is fine but even on the 3.6 log, i do not get any logical value for the corrected (SONR.Dist) height. 
Is the corrected height=height*(cos(roll)*cos(pitch))
Kind regards
[logs.zip](https://github.com/ArduPilot/ardupilot/files/337224/logs.zip)

rmackay9, 2016-09-08:
Copter-3.4 tilt compensates all range finders now.  That's not ideal for sonar but I think most are using lidar now anyway.  I'm going to close this if that's ok.

closed 2016-09-08 by rmackay9