ArduPilot/ardupilot #2384: Copter: guided should not climb above fence's alt limit

rmackay9, 2015-06-08:


rmackay9, 2015-06-08:
To be clear, it should act like Loiter and AltHold in that it should stop just below the fence.  In guided mode we can assume that the pilot is indirectly in control through the GCS and should be happy with the vehicle stopping just below the fence.

proficnc, 2015-06-08:
Thanks Randy.

salonijain12, 2016-04-08:
@rmackay9 , is the work related to this issue is still in progress, I can replicate this in SITL and I am interested in taking up this bug. Kindly advice.

rmackay9, 2016-04-08:
@salonijain12, yes, it's still an issue.  I don't think anyone is actively looking at it so it'd be great if you took a look at it!

salonijain12, 2016-04-19:
@rmackay9 , Initially I have disabled the fence and was flying in GUIDED mode. After reaching the desired point (given by "Fly To" command in SITL), I changed the mode to ALT_HOLD. After this, I noticed that my copter start descending and ALT_HOLD is not working as expected. Same is in case of mode LOITER.

I have gone through  http://ardupilot.org/copter/docs/altholdmode.html  (common problem #2), throttle is not maintained between 40-60% . Also the default value of THR_MID param is 500 (50%).

Is there anything which I am missing? I am not able to figure out, how to maintain throttle in GUIDED mode in SITL. Please suggest.

salonijain12, 2016-04-19:
@rmackay9 , Initially I have disabled the fence and was flying in GUIDED mode. After reaching the desired point (given by "Fly To" command in SITL), I changed the mode to ALT_HOLD. After this, I noticed that my copter start descending and ALT_HOLD is not working as expected. Same is in case of mode LOITER.

I have gone through http://ardupilot.org/copter/docs/altholdmode.html (common problem no. 2), throttle is not maintained between 40-60% . Also the default value of THR_MID param is 500 (50%).

Is there anything which I am missing? I am not able to figure out, how to maintain throttle in GUIDED mode in SITL. Please suggest.

salonijain12, 2016-04-19:
I am able to solve this using following steps:

In **STABILIZE** mode:

- arm throttle
- rc 3 1500

Then switch mode to **GUIDED**

- Perform Fly To operation
- Change mode to ALT_HOLD

After this the copter is not crashing and the desired Altitude is maintained.

Please suggest if these steps and behavior is as expected.

-

salonijain12, 2016-04-20:
@rmackay9 Thanks for the clarification of doubt.

Does stopping of guided mode controllers below the fence can be considered as if the copter has reached the destination waypoint just below the fence. That is considering the fence boundary as new destination waypoint and stopping it whenever it breaches the fence and if the copter eventually flies 100m outside the configured fence distance, switching the vehicle into LAND mode.
To be more specific:
	if (fence.get_breach_distance(new_breaches) <= AC_FENCE_GIVE_UP_DISTANCE) {
                    if (control_mode == GUIDED) {
                        
			//Updating the new stopping point as fence boundary
                        Vector3f stopping_point;
                        stopping_point.z = inertial_nav.get_altitude();
                        wp_nav.get_wp_stopping_point_xy(stopping_point);
 			wp_nav.set_wp_destination(stopping_point);
                        
                    } else if (!set_mode(RTL, MODE_REASON_FENCE_BREACH)) {
                        set_mode(LAND, MODE_REASON_FENCE_BREACH);
                    }
                }else{
                    // if more than 100m outside the fence just force a land
                    set_mode(LAND, MODE_REASON_FENCE_BREACH);
                }

I have tested this in SITL and it is working fine.
Please give suggestion/inputs on this approach.

rmackay9, 2016-08-29:
Bumping to AC-3.5.

rmackay9, 2017-02-15:
a quick update on this - in Guided mode's position controller, requests for the vehicle to fly outside the fence (horizontally and/or vertically) are rejected and a NAK is sent to the GCS.  I think this is correct behaviour.  What's not correct is the Guide mode's velocity controller which doesn't stop before it hits the fence.  I will update the subject line a bit to make this more clear.  Also bumping to AC3.6.

rmackay9, 2020-12-30:
I'm sure that guided mode's velocity controller now stops at the fence so closing

closed 2020-12-30 by rmackay9