ArduPilot/ardupilot #2390: change failsafe voltage/capacity in relation to distance from home.

geofrancis, 2015-06-09:
if you can work out roughly how many mah/km you multi-rotor does then the further you go from home the earlier it will engage rtl on battery fail-safe to avoid crashing on the way back because the battery gets too low.

or having 2 failsafe voltages? one is for RTL the second is for a fast land straight down to get it on the ground before it crashes.

rmackay9, 2015-06-09:
Related to this other issue: https://github.com/diydrones/ardupilot/issues/1759

rmackay9, 2015-06-09:
I'm amazed we didn't have this on the to-do list already.  It's certainly been requested but apparently not captured before.  The solo branch has this feature so we should be able to pull in for AC3.4.

ErikGroeneveld, 2015-06-09:
+1

volgende geschreven:

rmackay9, 2016-08-29:
Bumped to Copter-3.5.  We're just so late on AC-3.4 as it is.

geofrancis, 2017-02-14:
Has there been any progress on this?

WickedShell, 2018-03-10:
2 stage voltage/current is presented in #7213, it doesn't account for the distance from home, but does allow you to transition to land at a critical level. If that's sufficient then then this can be closed now since it's a duplicate of #1759

rmackay9, 2018-03-10:
I think this needs to stay open because the request is to failsafe to RTL at the optimal moment so that the vehicle barely makes it home.

WickedShell, 2018-03-10:
Yeah, that was my suspicion as well. That's not one I will be tackling in the super near future (or at least not yet in a manner that will help copter), but I think the issue itself is quite valid.

IamPete1, 2022-09-04:
I have implemented such a failsafe in scripting, its quite hard to get it "generic" in that case we used a lookup table from airspeed to power draw.