ArduPilot/ardupilot #2391: AP_Mission: store mission items as variable length chunks

magicrub, 2015-06-09:
Current, fixed size, scheme is wastes a lot of EEPROM/FRAM space and also does not even allow for all mavlink params to be stored. Since a lot of mission items contain unused space, this has the potential to store even more useful data while consuming less total memory.

cc @rmackay9 @tridge

WickedShell, 2015-06-09:
As long as we can export what the minimum/worst case scenario is. IE storing the most expensive waypoint type would allow X number of waypoints.

Also as a side note, while this is interesting, we would need a mechanism to inform the ground station how many way points could be stored. (Or teach the ground station how much space each waypoint type takes). IE right now I know that if I have an APM2 and upload more then 107 wp's I might be out of space and drop some, or on pixhawk if I have more then 700 waypoints I might drop some way points. At a mission planning stage this has to be accounted for, either in the GCS knowing how many waypoints can be stored, or in a clear mechanism to communicate that Y number of waypoints have been dropped. I have had multiple missions that simply didn't fit in the space, and since AP_Mission it has been gracefully, but silently handled, meaning that the only way to ensure the ground station knows what the mission looks like is to then read back the waypoints before every flight.

Actually, now that I've said that why isn't that the current behavior? If AP_Mission drops a waypoint for memory space then the GCS should be informed. (If this made it in within the last 6 months then I've just never noticed it)

rmackay9, 2015-06-10:
@WickedShell, I'm 99% sure that at least plane and copter correctly report back to the GCS a failure if they're unable to store the full list of mission commands received from the GCS.