ArduPilot/ardupilot #2402: EKF: integrate rangefinder

magicrub, 2015-06-10:
ALT_SOURCE is defined as:
// @Param: ALT_SOURCE
// @Description: This parameter controls which height sensor is used by the EKF. A value of will 0 cause it to always use baro altitude. A value of 1 will use the baro sensor, but switch to the range finder during optical flow navigation. A value of 2 will cause it to use GPS and switch to the baro sensor if GPS is not available.
// @Values: 0:Baro, 1:Range Finder, 2:GPS

Range Finder is poorly labeled here sice it's actually the sonar+opticalFlow. We should relabel value 1 to Optical flow and add "3: Range Finder"which uses AP_RangeFinder.

srobi, 2015-08-03:
"A value of 1 will use the baro sensor, but switch to the range finder during optical flow navigation."
Rangefinder maybe used outside of the optical flow navigation, purely for altitude measurement to support low altitude flight/terrain following with sensor.

magicrub, 2015-10-21:
After talking with @priseborough it was decided that this sort of thing should not be in the EKF but handled in the control loops where it already lives. Closing.

closed 2015-10-21 by magicrub