ArduPilot/ardupilot #2410: Copter/Plane: "CAM_TRIG_TIMER" parameter

robustini, 2015-06-12:
One nice feature is being able to take photos by delaying the shutter with a parameter.
Some cameras don't have this functionality in the software, i think it's simple to implement, and very usefull.
"CAM_TRIG_TIMER" it could be a settable parameter in second (with decimal unit), for example "1.5" takes a photo every 1.5 seconds.

iskess, 2015-06-13:
What would be a scenario where you would prefer an intervelometer over distance based triggering? 
I just thought of one, a copter is hovering and measuring change on the ground. It seems video would suffice in that example. For mapping, I can't think of any advantages of using an intervelometer.

robustini, 2015-06-13:
In some situations it can be very convenient.
Of course, i also use "CAM_TRIG_DIST" for mapping.

AndKe, 2016-03-04:
@robustini  - take a look at this https://github.com/diydrones/ardupilot/pull/3222   - when I made it, I thought it could could be used as intervalometer too, just set a very short CAM_TRIG_DISTANCE , and use MIN_INTERVAL to define time between.

Speaking about camera trigging: any idea what's my problem here ?  http://ardupilot.com/forum/viewtopic.php?f=111&t=15166

Naterater, 2019-01-07:
The delay is currently active as CAM_MIN_INTERVAL and is effective. I have personally tested it. http://ardupilot.org/plane/docs/parameters.html#cam-min-interval-minimum-time-between-photos

closed 2019-01-07 by Naterater