ArduPilot/ardupilot #2417: Builds are generating object files in working directory

magicrub, 2015-06-15:
Since the recent PX4 merge a "make px4-v2" will dump object files (*.o and *.d) into the working directory which makes things quite messy. They should do into a /obj or temp or somewhere other than the working directory and also cleaned correctly.

lucasdemarchi, 2015-06-15:
@magicrub what does "quite messy" look like?

magicrub, 2015-06-15:
my /ArduPlane folder triples in file count with every .cpp file now having a .cpp.d and .cpp.o file which then refreshes my eclipse folder browser. Also, searches done in Eclipse now include those files.

lucasdemarchi, 2015-06-15:
@magicrub ok, makes sense.

magicrub, 2015-12-08:
@guludo is this something in your realm of expertise?

lucasdemarchi, 2015-12-08:
@magicrub this is caused by PX4's build system.  It will eventually go away when waf is driving the PX4 build through its cmake integration.

lucasdemarchi, 2015-12-08:
@magicrub meanwhile you could also add exclusion filters in eclipse so it doesn't look for these files in the browser or search

magicrub, 2015-12-08:
Got it, thanks.

guludo, 2015-12-09:
@magicrub, sorry for the delayed answer. But as @lucasdemarchi said, that should be fixed when PX4 build is integrated in waf. :-)

OXINARF, 2017-06-04:
@magicrub Do you still want this open? waf build system uses a build folder for all produced files, but Make still pollutes the source dir.

magicrub, 2017-06-04:
This should stay open as long as we support make

khancyr, 2019-04-05:
make isn't support anymore, closing

closed 2019-04-05 by khancyr