ArduPilot/ardupilot #2420: Copter accepts mission_item_encode as a guided command

hamishwillee, 2015-06-16:
I ran the following code in DroneKit-Python in guided mode - it successfully sets the Yaw. According to @rmackay9 the mission_item_encode should only be for mission items in missions so this should not work. If you see the line `` 2, # current - set to 2 to make it a guided command`` it looks like at some point this method was overloaded so it could be used in guided mode.

I don't know how to fix this, because presumably some people might be relying on it - mark as deprecated somehow?

Anyway, the yaw command can also be send in guided using command_long_encode, and that works fine. The only "issue" is that for command long you can't set a frame, which may be inconvenient.

```
def condition_yaw_1(heading, relative=False):
    """
	Send condition_yaw mavlink command so vehicle points at specified heading (in degrees).
    By default the heading is absolute, but you can set the relative parameter to True to make 
    yaw relative to the direction of travel.
    """
    if relative:
        is_relative=1 #yaw relative to direction of travel
    else:
        is_relative=0 #yaw is an absolute angle
    # create the CONDITION_YAW command
    msg = vehicle.message_factory.mission_item_encode(0, 0,  # target system, target component
            0,     # sequence
            mavutil.mavlink.MAV_FRAME_GLOBAL_RELATIVE_ALT, # frame
            mavutil.mavlink.MAV_CMD_CONDITION_YAW,         # command
            2, # current - set to 2 to make it a guided command
            0, # auto continue
            heading,    # param 1, yaw in degrees
            0,          # param 2, yaw speed deg/s
            1,          # param 3, direction -1 ccw, 1 cw
            is_relative,          # param 4, relative offset 1, absolute angle 0
            0, 0, 0)    # param 5 ~ 7 not used
    # send command to vehicle
    vehicle.send_mavlink(msg)
    vehicle.flush()
```

OXINARF, 2017-06-02:
Although I agree that while in Guided mode we shouldn't accept waypoints and yaw commands inside mission items, even MAVProxy still relies on that. We should encourage people to use other messages for Guided usage, but I don't think we can safely remove this hack without breaking support for some people. Even in the last few days this was talked as solution in Gitter so there may be people starting using it now.

For this reason, I'm closing this. Feel free to re-open it if you think differently.

closed 2017-06-02 by OXINARF

hamishwillee, 2017-06-02:
Fair enough. If everyone does it then it is a feature, not a bug :-)