ArduPilot/ardupilot #2421: Storm32: integrated camera trigger does not work when connected via MavLink

hsteinhaus, 2015-06-16:
Regarding to http://www.olliw.eu/storm32bgc-wiki/Serial_Communication#Comments_to_STorM32_specific_Mavlink_commands , the Storm32 gimbal controller should be able to trigger the camera with it's built-in IR trigger via MavLink message MAVLINK_MSG_ID_DIGICAM_CONTROL.  AP_Camera seems to be aware of MAV_CMD_DO_DIGICAM_CONTROL as well. 

May be for some reason the appropriate message is not forwarded to the link to which the Storm32 gimbal is connected?

SovGVD, 2015-12-11:
AdruCopter 3.3.2 (self build with NMEA fix for GPS/GLONASS from Alexey Kozin).

I made some tests with MissionPlanner and do mp-storm32-pano-example1.py (MAV.doCommand(MAVLink.MAV_CMD.DO_DIGICAM_CONTROL, 0, 0, 0, 0, 1, 0, 0);)
1) connect Storm32 via UART (FTDI clone adapter) - shutter OK
2) connect Storm32 to Pixhawk and choose HEXACOPTER-1 in SysID window - shutter FAILED
3) connect Storm32 to Pixhawk and choose STORM32-xx in SysID window - shutter OK

Looks like Storm32 do the job great, but something wrong with AP_Camera library or some issue with mavlink routing or "broadcasting".

Any suggestions?

rmackay9, 2015-12-12:
As of a couple of days ago, OlliW was still working on the support in the SToRM32 gimbal.  I'm not aware of anything that needs changing on the ardupilot side but if there's is something .. if someone's got the details I can have a look.

SovGVD, 2015-12-12:
It's actually hard to understand code "on the fly", so if you have a spare time, could you please explain string `if ((routes[i].sysid == mavlink_system.sysid) && !sent_to_chan[routes[i].channel]) {` in MAVLink_routing::send_to_components.
Is this code could block camera trigger command for mavlinked storm32? What value for `mavlink_system.sysid`? And what is `routes[i].sysid` when `i` index reached gimbal route?

SovGVD, 2015-12-12:
Ok. Finally. Mega-super-crazy-lazy hack (sorry for not a diff)... I think it could...must be better, but I need to finish my copters ASAP, so... sorry =)

AP_Camera.h into private section
```    bool _gimbalcam_search;              // try to search gimbal
    bool _gimbalcam_initialised;              // true once the driver has been initialised
    uint8_t _gimbalcam_sysid;                 // sysid of gimbal
    uint8_t _gimbalcam_compid;                // component id of gimbal
    mavlink_channel_t _gimbalcam_chan;        // mavlink channel used to communicate with gimbal.
```
AP_Camera.cpp into trigger_pic after GCS_MAVLINK::send_to_components(&msg);:
```	if (!_gimbalcam_search) {
		_gimbalcam_search=true;
		if (GCS_MAVLINK::find_by_mavtype(MAV_TYPE_GIMBAL, _gimbalcam_sysid, _gimbalcam_compid, _gimbalcam_chan)) {
		    _gimbalcam_initialised=true;
		}
	}

	if (_gimbalcam_initialised) {
		mavlink_msg_command_long_send(_gimbalcam_chan,
                                  _gimbalcam_sysid,
                                  _gimbalcam_compid,
                                  MAV_CMD_DO_DIGICAM_CONTROL,
                                  0,
                                  0,
                                  0,
                                  0,
                                  0, 1, 0,
                                  0);
	}
```

ArduCopter 3.3.2 + storm32 0.87e

olliw42, 2015-12-13:
"As of a couple of days ago, OlliW was still working on the support in the SToRM32 gimbal."

this is a misunderstanding, the thread which might have given that impression is 8 months old :). http://www.rcgroups.com/forums/showthread.php?t=2392046&pp=50#post33407542
http://www.rcgroups.com/forums/showthread.php?t=2494532&page=2&pp=50#post33339030

it doesn't belong to here, but since I now happen to be here I'd like to mention, there seems to be also some communication issues with Digaus' app which - I think - uses Drone Kit: http://www.rcgroups.com/forums/showpost.php?p=33383706&postcount=128, just to have mentioned it

:)

SovGVD, 2015-12-13:
Looks like `void MAVLink_routing::send_to_components(const mavlink_message_t* msg)` send messages only "inside" of the copter (sysid=1). Also I try change `if ((routes[i].sysid == mavlink_system.sysid) && !sent_to_chan[routes[i].channel]) {` to `if ((routes[i].sysid == mavlink_system.sysid || routes[i].mavtype ==MAV_TYPE_GIMBAL) && !sent_to_chan[routes[i].channel]) {` but without success.

So now the only way I found is send command direct to gimbal in AP_Camera:
```
mavlink_msg_command_long_send(_gimbalcam_chan,
                                  _gimbalcam_sysid,
                                  _gimbalcam_compid,
```
But it might be better to move code to AP_Mount_* and call gimbal methods inside AP_Camera.

rmackay9, 2015-12-14:
My guess is that the issue is that the gimbal isn't setup to use the same system id as the vehicle but with a different component id.  That's really how it should be done because the gimbal is a component of the vehicle.  The only problem then is that I think some of the ground stations may become confused because they're not correctly separating message by the sysid+component id combination.  Still, the correct fix would be to get the GCS issues resolved.

SovGVD, 2015-12-14:
So if I set sysid=1 for storm32 it will work without any quick and dirty hacks? Will try today.

rmackay9, 2016-01-27:
Ok, I think we've found the issue and will pull this into Copter-3.3.3-rc2 which will go out for testing in about a week.  https://github.com/diydrones/ardupilot/commit/cefb32566f043e9b7b94c58b07ed50afe897f380

The issue is already fixed in master so if you want to test immediately, you could load up "Copter-3.4-dev" from the mission planner by going to the Install Firmware screen and pressing "Ctrl-Q" and then click on the appropriate vehicle icon to start the install.  Firmware loaded like this is our development firmware though so, although it's probably fine, it hasn't been thoroughly tested yet.

SovGVD, 2016-01-30:
Thank you. I will try it later. Any way I need to build firmware with my own fixes for NMEA GPS+GLONASS device (https://github.com/diydrones/ardupilot/issues/1241).

geekness, 2016-03-04:
Did this get resolved, and has it been integrated into the latest stable version of copter? 
Are there any settings that need to be modified/changed in MP to make this work, like do I HAVE to set a channel to trigger the shutter or can I leave it blank? I manually trigger the IR and the pitch by separate PWM cables from Rx to storm32.

hsteinhaus, 2018-03-31:
Many successful photo flights with Storm32 and MAVLink control (MNT_TYPE=4) and camera triggered via Storm32 in between - closing.

closed 2018-03-31 by hsteinhaus