ArduPilot/ardupilot #2424: Add pre-arm check for PID sanity

sgofferj, 2015-06-16:
As written [on DIYD] (http://diydrones.com/profiles/blogs/on-the-importance-of-being-thorough-when-setting-up-your-uav), I had a mild copter crash due to me inadvertently setting all THR_* PIDs to zero. In this context, Randy suggested a basic PID sanity pre-arm check on the dev list to make sure, important PIDS do make some basic sense.
Not sure how much sense that would make for Plane and Rover but for Copter and Tradheli, it probably would make lots of sense.

rmackay9, 2016-08-29:
Bumping to AC-3.5.  We're simply out of time for AC-3.4.

lthall, 2017-11-13:
I don't think you should include the D term as some of the Copter aircraft don't use them. If the FF is set then you should ignore the PID too.

closed 2018-01-12 by lthall