ArduPilot/ardupilot #2426: Storm32 with MavLink: Gimbal moves to a position outside of the configured range

hsteinhaus, 2015-06-16:
When powering on the copter without switching on the TX before, the gimbal tries to move into an extreme position outside the range defined by MNT_*_MIN/MAX parameters. This behavior could cause damage to the gimbal and/or the copter.

lvale, 2015-06-24:
isn't this very similar to issue https://github.com/diydrones/ardupilot/issues/2375?

hsteinhaus, 2015-06-25:
#2375 is probably the reason for the movement, but the main subject of this bug is that the gimbal driver does not respect it's configured mechanical limits.

lvale, 2015-06-25:
@hsteinhaus 
I believe that also having the connection type (serial vs PWM) between the gimbal and the PixHawk, and also which firmware and the parameters are on the SToRM32,  would be good to troubleshoot this. I have also the STorM32 plugged via serial and had no issues (yet). Only possible difference that I might see is that I have the power to the gimbal on a normal switch, that is normally off.

walmis, 2015-07-16:
This patch should fix this bug.

```
diff --git a/libraries/AP_Mount/AP_Mount_Backend.cpp b/libraries/AP_Mount/AP_Mount_Backend.cpp
index 08f35de..cde87f3 100644
--- a/libraries/AP_Mount/AP_Mount_Backend.cpp
+++ b/libraries/AP_Mount/AP_Mount_Backend.cpp
@@ -122,7 +122,10 @@
 // returns the angle (degrees*100) that the RC_Channel input is receiving
 int32_t AP_Mount_Backend::angle_input(RC_Channel* rc, int16_t angle_min, int16_t angle_max)
 {
-    return (rc->get_reverse() ? -1 : 1) * (rc->radio_in - rc->radio_min) * (int32_t)(angle_max - angle_min) / (rc->radio_max - rc->radio_min) + (rc->get_reverse() ? angle_max : angle_min);
+    if(rc->radio_in < rc->radio_min || rc->radio_in > rc->radio_max) {
+    	return 0;
+    }
+	return (rc->get_reverse() ? -1 : 1) * (rc->radio_in - rc->radio_min) * (int32_t)(angle_max - angle_min) / (rc->radio_max - rc->radio_min) + (rc->get_reverse() ? angle_max : angle_min);
 }
 
 // returns the angle (radians) that the RC_Channel input is receiving
```

Naterater, 2019-01-07:
Could this be solved by @rmackay9's recent PR?

rmackay9, 2019-01-07:
Yes, fixed now in stable Copter.

closed 2019-01-07 by rmackay9