ArduPilot/ardupilot #2427: Need mechanism in Copter to reset yaw to follow vehicle heading in guided mode

hamishwillee, 2015-06-17:
By default when you enter guided mode, the yaw of the vehicle follows the vehicle heading. If you change the yaw using [MAV_CMD_CONDITION_YAW](http://copter.ardupilot.com/wiki/common-planning-a-mission-with-waypoints-and-events/common-mavlink-mission-command-messages-mav_cmd/#mav_cmd_condition_yaw) you will have to control the yaw manually for the rest of the script, because there is no way good way to reset the yaw back to following vehicle heading (actually, you can toggle the mode away from GUIDED and back again, but then RC Controller settings for throttle etc may affect vehicle behaviour - so this is not acceptable).

A desirable implementation is to update [MAV_CMD_CONDITION_YAW](http://copter.ardupilot.com/wiki/common-planning-a-mission-with-waypoints-and-events/common-mavlink-mission-command-messages-mav_cmd/#mav_cmd_condition_yaw) param4 to have a third option that sets the yaw as relative to the vehicle heading (in addition to the absolute values and relative to the last yaw setting). Setting this and a heading of zero would emulate the default behaviour. Setting this and a heading of anything else would allow a yaw to be specified relative to the vehicle's direction of travel.

A less desirable but still good implementation would be to simply reset to the default behaviour, perhaps using Param 5.

Guohao91, 2020-05-28:
@hamishwillee 
Hi. Wonder if the raw command "mavutil.mavlink.MAV_CMD_CONDITION_YAW" can be used under NoGPS mode?

hamishwillee, 2020-06-01:
@Guohao91 No idea. Probably - it requires a heading but not a position, and that comes from the compass.

rmackay9, 2020-06-01:
@Guohao91, by "NoGPS" mode do you mean while using the mode called "Guided_NoGPS" or do you mean while doing non-GPS position estimation (using one of these methods: https://ardupilot.org/copter/docs/common-non-gps-navigation-landing-page.html).

Guided_NoGPS won't work with the condition-yaw command.  Auto and Guided modes should accept the condition-yaw I think.

Guohao91, 2020-06-01:
@rmackay9 I mean try to call condition-yaw command while using the mode "Guided_NOGPS". Anyways, thanks your answer!

rmackay9, 2020-06-02:
@Guohao91, ok, so as mentioned above, the condition-yaw command won't work with Guided_NoGPS.  Still, to use Guided_NoGPS you'll need to send in [set-attitude-target mavlink messages](https://mavlink.io/en/messages/common.html#SET_ATTITUDE_TARGET) which has a "body_yaw_rate" and "q" field which can be used to set the earth-frame yaw angle or z-axis rotation rate which is the same (I think) as what the condition-yaw command supports.

Guohao91, 2020-06-02:
@rmackay9  Thanks a lot Mr.Randy! Yeah I am actually using "set-attitude-target" mavlink messages currently. I just found condition-yaw is more straight so I gave a try to ask.
well now, I am not so sure: does set-attitude-target mavlink messages recognize minus yaw angle? i.e. I typed : set_attitude(yaw_angle = -25, duration = 2). does this functionlly work?

malfarizit, 2020-10-10:
> @rmackay9 Thanks a lot Mr.Randy! Yeah I am actually using "set-attitude-target" mavlink messages currently. I just found condition-yaw is more straight so I gave a try to ask.
> well now, I am not so [...]

yap u can use set attitude like this for guided_nogps.

malfarizit, 2020-10-10:
anyone can change altitude on guided_nogps?