ArduPilot/ardupilot #2445: mission index should reset on mission update

magicrub, 2015-06-23:
When uploading a new mission while in auto the current mission stays in RAM and continues to execute. With MISSION_RESTART=0, if you switch to manual then back to auto you will continue executing the RAM mission, not the new mission in flash. To execute the new mission you must reset mission index to 1.

This may seem obvious but consider this scenario:
- boot /w some unknown previously uploaded mission.
- switch to auto
- switch back to manual
- upload new mission
- switch to auto and launch.

You had stayed on mission cmd index=1 the whole time. You will launch with the old mission cmd 1 instead of your new mission cmd 1.

On a mission upload, reset index to 1 will solve this problem

rmackay9, 2015-06-23:
Sounds like we have a bit of a problem.
I guess we need to be careful that when we fix this that the uploading of partial missions still works ok.  So for example if the user uploads waypoints that haven't been run yet.. we wouldn't want to interrupt the mission.

magicrub, 2015-06-23:
I already do detection of full mission updates when I dump the new mission
to DF, the same mechanism could trigger a mission restart.

I had done a half-fix to test this by adding logic in the reset-or-resume
function to reset if the current index was 1. That fixes my particular
problem but it is not a very good general solution. The above solution is
better.

tridge, 2015-06-23:
I would prefer not having an auto-update. I think instead the GCS should pop up a dialog to ask if the mission should be restarted?

magicrub, 2015-06-23:
After talking it over with @tridge , @rmackay9  we think a reset of the current mission item if ok only when not currently executing it. That is, when switching from non-AUTO to AUTO you should always clear the cached copy and update it with the mission item in flash memory.

magicrub, 2015-06-23:
fixed via https://github.com/diydrones/ardupilot/pull/2448

closed 2015-06-23 by magicrub