ArduPilot/ardupilot #2447: Copter: Add support for MAV_FRAME_BODY_NED in SET_POSITION_TARGET_LOCAL_NED

hamishwillee, 2015-06-23:
[SET_POSITION_TARGET_LOCAL_NED](https://pixhawk.ethz.ch/mavlink/#SET_POSITION_TARGET_LOCAL_NED) allows you to send commands for position and velocity control. 

The message definition supports a number of frames: MAV_FRAME_LOCAL_NED = 1, MAV_FRAME_LOCAL_OFFSET_NED = 7, MAV_FRAME_BODY_NED = 8, MAV_FRAME_BODY_OFFSET_NED. 
However as far as I can tell it doesn't matter what you use, parameters are interpreted as relative to the home location.

I don't actually understand what the frames really mean! But:

* For position control, it would be very useful if we could have a frame that allows us to set position relative to the vehicle - ie "go 10m north and 20 metres west" rather than having to translate decimal degrees. Having variants where the altitude is relative to the current vehicle position or above ground level would also be useful
* For velocity control I'm not so sure. I imagine though it would be useful to be able to apply velocity vectors aligned with the front of the vehicle. I'm thinking "ease of defining a straffing path". 

The velocity idea is is theoretical, the dronekit examples haven't needed it yet. The position case is a real problem, for which I ended up having to write my own translation code.

MrDadaGuy, 2015-07-14:
I also have this same need, and I am also writing my own translation code in the absence of the feature.

TShapinsky, 2015-08-10:
I ended up writing my own translational code as-well because this does not work.

hamishwillee, 2015-08-11:
Just FYI, we added some translation code in [dronekit docs here](http://python.dronekit.io/guide/copter/guided_mode.html) and will probably add this to a common library at some point. However it would be better if the autopilot handled this as per the protocol.

rmackay9, 2015-08-11:
I think I've done this and it seems to be working in SITL so I plan to look at it a bit more and push to master: https://github.com/rmackay9/rmackay9-ardupilot/commits/guided-landed2

rmackay9, 2015-08-11:
ok this is in master now so closing this issue.  It likely won't be released with AC3.3 but will go out with AC3.4.

closed 2015-08-11 by rmackay9

hamishwillee, 2015-08-12:
@rmackay9 Thanks very much. 

[Here you translate the velocity vector between frames](https://github.com/rmackay9/rmackay9-ardupilot/blob/231461abd602326ea823eed45a9859e809aa3399/ArduCopter/GCS_Mavlink.cpp#L1498). Can you explain the difference - I think that in the local frame vx, vy, vz are NED, but in the body frame they are relative to the axes of the body (ie if the vehicle was on its side, then the "D" would follow the vehicle axis, not gravity)

rmackay9, 2015-08-12:
@hamishwillee,
In the way I've implemented it, the only difference is the heading.  So for example, if you ask for (1,0,0) velocity with NED it will fly north at 1m/s.  In BODY the vehicle will fly forward at 1m/s.

hamishwillee, 2015-08-13:
@rmackay9 Thanks, I think I need to get a bit more information:

For velocity:

> In the way I've implemented it, the only difference is the heading. So for example, if you ask for (1,0,0) velocity with NED it will fly north at 1m/s. In BODY the vehicle will fly forward at 1m/

 What happens if the value is (0,1,0)? Will it fly sideways at 1 m/s, and if so in what direction relative to the front of the vehicle (ie if I was looking straight at the vehicle, which way would it move - left or right?

For the positions I found the [frames explanation](https://github.com/mavlink/mavlink/blob/master/message_definitions/v1.0/common.xml#L456) confusing. I *think* I understand:

* MAV_FRAME_LOCAL_NED - specified x,y, z in NED frame relative to home position
* MAV_FRAME_BODY_NED - specified x,y, z in NED frame relative to current position

but what do the offsets do? MAV_FRAME_LOCAL_OFFSET_NED  and MAV_FRAME_BODY_OFFSET_NED

rmackay9, 2015-08-13:
@hamishwillee,
(0,1,0) will make it fly to the right (or east) at 1m/s.

"NED" means north-east-down of course, so I'm a little confused how "Body" could have "NED".  What's north and east doesn't seem it should change depending upon the orientation of the vehicle.  In any case, I've interpreted any frame description with "BODY" in it as meaning it's in relation to the vehicle's current heading.

For "OFFSET", I think this is only useful for position requests.  So if "OFFSET" appears in the frame then the position is from the vehicle's current position.  If "OFFSET" is not part of the Frame's description then it's in relation to home.  We ignore the "OFFSET" component for velocity request.  I had thought that perhaps we could interpret it as a request to increase the velocity by the request.. but that seemed a bit confusing.

hamishwillee, 2015-08-13:
@rmackay9 Sorry, the "or east" in the first line makes this ambiguous. I think what you are saying is that in local frame (0,1,0) will go east. In the body frame you will move right (where right imagines you are sitting in the UAV cockpit).Is that correct? 

> "NED" means north-east-down of course, so I'm a little confused how "Body" could have "NED". What's north and east doesn't seem it should change depending upon the orientation of the vehicle. In any case,

Agree it is confusing and the spec does not help. We probably need a voice conversation about this.

So what people want to be able to do is

1.  For position requests, be able to specify the new position relative to their current position in metres in NED frame - 10 metres North, 6 metres east, 2 metres down of where I am now
1.  For position requests, be able to specify the new position relative to their current position in metres *and relative to the current heading*   - 2 metres forward, 3 metres to the left, 3 metres up.
1. For velocity requests - velocity forward, to right, to down.

So how does that map to the different frames?

rmackay9, 2015-08-14:
1. MAV_FRAME_LOCAL_OFFSET_NED
2. MAV_FRAME_BODY_OFFSET_NED
3. MAV_FRAME_BODY_NED or MAV_FRAME_BODY_OFFSET_NED

hamishwillee, 2015-08-14:
Thanks @rmackay9 I think this is now understood. Can you please confirm http://dev.ardupilot.com/wiki/copter-commands-in-guided-mode/#set_position_target_local_ned is correct/ok description?

rmackay9, 2015-08-14:
looks wonderful.  By the way Freddia told me that what confused them when first using the command was the type_mask bitmask.  In particular the issue was that Copter expects the user will provide a 3D vector for position and/or velocity.  They originally tried to provide just one axis of the 3D vector.  So if they wanted the vehicle of fly north, they'd provide just the "vel_x" value and set just the single appropriate bit in the type_mask bitmask.
I could change the behaviour in the Copter code but it adds a fair bit of complexity that I don't think is probably worth it.  Maybe we should just add a bit of detail on how people should set that type_mask field.  Maybe also provide the actual numbers for setting the bits in case the user is providing a position or a velocity.

hamishwillee, 2015-08-17:
Hi @rmackay9 
That sounds sensible to me.

To do this I have had to import the definition and from the protocol into the wiki and then added a note in the type_mask field explaining how it is used. Can you please check [set_position_target_local_ned](http://dev.ardupilot.com/wiki/copter-commands-in-guided-mode/#set_position_target_local_ned) again now

As users of the global frame version will have the same problem, I've updated that too:[set_position_target_global_int](http://dev.ardupilot.com/wiki/copter-commands-in-guided-mode/#set_position_target_global_int). Can you please confirm that I got the specific frame right for vx etc (e.g. X velocity in **MAV_FRAME_LOCAL_NED** frame in meter/s). 
Also, the co-ordinate frame follows the spec: "Valid options are: MAV_FRAME_GLOBAL_INT, MAV_FRAME_GLOBAL_RELATIVE_ALT_INT, MAV_FRAME_GLOBAL_TERRAIN_ALT_INT" but I suspect on copter we only support the first two frames?

Nearly there!

rmackay9, 2015-08-17:
@hamishwillee,
yes, the set-pos-target-local-ned looks good.
yes, the set-position-target-global-int also looks good although we only support the first two (i.e. we don't support mav-frame-global-terrain-alt-in)
thanks for this!

hamishwillee, 2015-08-18:
Hi @rmackay9 Great - you're welcome. I've removed MAV_FRAME_GLOBAL_TERRAIN_ALT_INT.

We're done!

flightzoomer, 2015-10-11:
Hi Randy

I just checked tag Copter-3.3 and I can see there the code mentioned above ( https://github.com/rmackay9/rmackay9-ardupilot/blob/231461abd602326ea823eed45a9859e809aa3399/ArduCopter/GCS_Mavlink.cpp#L1498 ).

Does this not mean that the different frames are supported in 3.3 already? Or am I missing something (really stupid probably)?

Thanks&regards
Martin

hamishwillee, 2015-10-11:
@rmackay9 That's a good question - my docs say this went in in 3.4 https://github.com/diydrones/ardupilot/blob/Copter-3.3/ArduCopter/GCS_Mavlink.cpp#L1498

rmackay9, 2015-10-12:
@flightzoomer is right, it went out with Copter-3.3.  I wasn't intending on that originally but then we ended up doing a rebase on master somewhere around -rc9-ish... sorry for not circling back but there's just so many little things..

hamishwillee, 2015-10-12:
@rmackay9 Thanks Randy. I've updated the note [in wiki docs](http://dev.ardupilot.com/wiki/copter-commands-in-guided-mode/#set_position_target_local_ned). As you say, lots to keep keep track of.

flightzoomer, 2015-10-12:
Thank you guys! I know very well that software development works that way and 100% in anything is not possible in reality... I anyway appreciate your diligent work.

jlian, 2015-10-29:
@rmackay9 @hamishwillee I just did tests with this command via Copter 3.3 and DroneKit-Python and everything went smoothly and as expected (just simple going forward and backwards). Didn't realize that this was so recent!

rmackay9, 2015-10-29:
@jlian, ah, great txs for the feedback!

DarylWM, 2016-02-28:
Thanks for your work on this @rmackay9. I tried MAV_FRAME_BODY_NED in SITL (APM:Copter V3.3-dev) and [the vehicle movement didn't seem to conform to body frame](https://discuss.dronekit.io/t/mav-frame-body-ned-not-working-as-expected/277/1). Have I misunderstood how it works?

hamishwillee, 2016-02-28:
@rmackay9 His code looks fine and matches the description. Either this isn't working quite right OR the heading is the direction that the vehicle is currently moving and does not match the yaw direction ... and defaulting to North on first use. Can you respond here? https://discuss.dronekit.io/t/mav-frame-body-ned-not-working-as-expected/277

shubham-shahh, 2021-11-09:
Hi, @rmackay9 @hamishwillee, when I send the message SET_POSITION_TARGET_LOCAL_NED with only velocities for example to the right, the drone's heading moves to the right as well which is not intended

is there a way to give only velocities and keep the heading of the UAV constant