ArduPilot/ardupilot #2466: Support for BlinkM I2C LED for use as Nav/Status Lights

sandysoundy, 2015-06-28:
Hello everyone, absolutely love and appreciate your work! 
This may not be a new request but it would be great to enable the use of BlinkM LED's from the pixhawk I2C port to add some nav/status lights to the aircraft on each arm of the copter to use as a reference to the status/direction of the machine.

Cheers,

Sandy

rmackay9, 2015-06-29:
In case some developer wants to take this on, there is already a blinkm driver in the PX4Firmware layer so the first step would be to test that works.  Then we need to make an ardupilot driver in the AP_Notify directory perhaps similar to the ToshibaLED driver.

So overall it's not terribly hard.  If you email drones-discuss@googlegroups.com and we can give more detailed advice on the programming side.

sandysoundy, 2015-06-30:
> <reply+00c7b0d850ccb8c573153c26f5ad80cc80ecc13c4e472b9a92cf0000000111a867949
> 2a169ce0575b6be@reply.github.com>
> Lights (#2466)

Hi Randy,

Thanks for the encouragement!
I tried to email drones-discuss@googlegroups.com but I got a failure
delivery status notification Do I need to join the group before I can post?

Regards,

Sandy

davidbuzz, 2015-12-26:
BlinkM is generally considered inferior to the ToshibaLEDs, as there are licensing issues preventing it's use for any commercial purpose ( look at the "firmware license" link here: http://thingm.com/products/blinkm/  ), and it's price-per-unit is around double other ToshibaLED compatible units.

auturgy, 2019-11-18:
with ws2812 now implemented, I'm closing this as overtaken by events.  BlinkM led's aren't really relevant anymore.

closed 2019-11-18 by auturgy