ArduPilot/ardupilot #2470: Copter 3.3 rally point priority to home wp error event if home is closer

roque-canales, 2015-06-29:
If drone is closer to home point than set rally point, if rtl invoke, drone go to rally point....

rmackay9, 2015-06-29:
So this request is to exclude home from the list of possible landing location if rally points are used?

This makes sense if you were taking off from a boat and wanted the RTL to go to the shore instead.

We could add a RALLY_BEHAVE parameter which allows control of whether home is included or excluded from the list of acceptable landing locations if rally points are used.

roque-canales, 2015-07-28:
Randy, sorry but for me its an error if drone no use standard home point when rally point is used...
example:
Rally point set at 500meters from home
if drone is above home point, if rtl invoke drone go to rally point...

I like the idea of rally behave to set if we want to use or not home point when rally p is set.

Could you add this on the next rc?

rmackay9, 2015-07-28:
I've realised that this can actually be done by setting the RALLY_LIMIT_KM parameter to zero.
The description says this although the final sentence is a bit misleading.

// @Description: Maximum distance to rally point. If the closest rally point is more than this number of kilometers from the current position and the home location is closer than any of the rally points from the current position then do RTL to home rather than to the closest rally point. This prevents a leftover rally point from a different airfield being used accidentally. If this is set to 0 then the closest rally point is always used.

rmackay9, 2015-07-28:
So perhaps what we could do is test that I'm right and if so, default the value of this parameter to zero for Copter.

KiwiHC16, 2015-07-29:
From: http://copter.ardupilot.com/wiki/common-rally-points/ <http://copter.ardupilot.com/wiki/common-rally-points/>

RALLY_LIMIT_KM <http://plane.ardupilot.com/wiki/arduplane-parameters/#Rally_Limit_PlaneRALLY_LIMIT_KM> is the maximum distance a Rally Point may be from the aircraft to be considered for an RTL event.  If all Rally Points are greater than this distance from the aircraft, then the Home location is used for RTL events (at altitude ALT_HOLD_RTL <http://plane.ardupilot.com/wiki/arduplane-parameters/#RTL_altitude_PlaneALT_HOLD_RTL>) unless Home is farther away than the nearest Rally Point — in which case the nearest Rally point is used.  This parameter is to prevent fly offs if Rally Points have been specified for multiple flying fields. This parameter can be disabled if set to 0.

KiwiHC16, 2015-07-29:
Result of some SITL Tests:
- if RALLY_LIMIT_KM set to 0, Copter always goes to a Rally point when RTL is triggered, never to Home even if very close to it.
- if RALLY_LIMIT_KM set to a large number to include all Rally, Copter always goes to Rally point and never to Home even if very close
- if RALLY_LIMIT_KM set to a small value 0.01 (10m), Copter goes back Home excepts if a Rally point is closer.
So we could say, Copter goes back home only if there is no Rally point in the area. Otherwise it always goes to Rally point.
Simulation in line with definition as far as I can see.

KiwiHC16, 2015-07-29:
What about RALLY_HOME_EXCLUDED set to 0 as  default (Home is used as a Rally point), could be set to 1 if you launch from a boat.

rmackay9, 2015-07-29:
@KiwiHC16, thanks a lot for doing that testing.  That's a real help.
So I guess the short-term work around for @roque-canales is to set the RALLY_LIMIT_KM to a very small number.  A more intuitive interface would be like you say, a new parameter to include/exclude home.  At the moment our params need to be no more than 16 characters but in any case, it's a good idea for sure.

KiwiHC16, 2015-07-30:
I guess the summary is in this row of code from  AP_Rally.cpp:
min_dis = distance to closest Rally point

if (       (_rally_limit_km > 0) 
     && (min_dis > _rally_limit_km*1000.0f) 
     && (get_distance(current_loc, home_loc) < min_dis)) {
        return false; // use home position
    }

KiwiHC16, 2015-07-30:
Before trying to change any code, I'm wondering if we are not mixing 2 concepts. One is RTL (return to home) and one is "Go to a Safe Location to Land as soon as possible". If I do a mission, I guess I would tend to use RTL as the last mission command. If Rally are defined, the copter should go to a rally point and not to launch point. If rally are defined then I should set a WP to Home location, put this WP as the last WP in the mission and then land. Perhaps we should have 2 modes: one is RTL = return to home/launch and one one GoToRally (as a safety feature). What do you think ?

rmackay9, 2015-07-31:
@KiwiHC16, I think separating RTL and Rally points will just make it more confusing to the user.  Rally points are also so rarely used from what I've seen that we should try and keep it simple.

Thanks for finding the line in the code.  Another option instead of adding a new parameter would be to have two special values.  So "0" could be changed to mean, "always include home" and >10000km could mean "never include home".

KiwiHC16, 2015-07-31:
I think I found the solution to stay very simple without touching anything. When a RTL is triggered then go  to a Rally point if there is a FS situation otherwise we go back home.

In the code the solution is limited to minimum in control_rtl.cpp:

Before:
#if AC_RALLY == ENABLED
    // rally_point.alt will be the altitude of the nearest rally point or the RTL_ALT. uses absolute altitudes
    Location rally_point = rally.calc_best_rally_or_home_location(current_loc, get_RTL_alt()+ahrs.get_home().alt);
    rally_point.alt -= ahrs.get_home().alt; // convert to altitude above home
    rally_point.alt = max(rally_point.alt, current_loc.alt);    // ensure we do not descend before reaching home
    destination.z = pv_alt_above_origin(rally_point.alt);
#else
    destination.z = pv_alt_above_origin(get_RTL_alt());
#endif

After:
#if AC_RALLY == ENABLED
    // If in FailSafe situation go to a Rally point otherwise Return to launch: RTL
    if ( failsafe.battery || failsafe.radio ) {
    // rally_point.alt will be the altitude of the nearest rally point or the RTL_ALT. uses absolute altitudes
    Location rally_point = rally.calc_best_rally_or_home_location(current_loc, get_RTL_alt()+ahrs.get_home().alt);
    rally_point.alt -= ahrs.get_home().alt; // convert to altitude above home
    rally_point.alt = max(rally_point.alt, current_loc.alt);    // ensure we do not descend before reaching home
    destination.z = pv_alt_above_origin(rally_point.alt);
    }
    else {
    destination.z = pv_alt_above_origin(get_RTL_alt());
    }
#else
    destination.z = pv_alt_above_origin(get_RTL_alt());
#endif

It does compile and I need to test if it works. Need to find a way to simulate a low battery and a loss of radio. Anybody knows how to do that ?

KiwiHC16, 2015-07-31:
Just did tests with SIM_RC_FAIL and SIM_BATT_BATT to trigger FS. Works as expected. Tested RTL request and came back home as expected. Will do more tests and then propose the correction/enhancement to be included.

KiwiHC16, 2015-08-01:
I just created a pull request #2637. I'm working on a second one to come soon. (I hope the pull is right, it's my first pull request).

KiwiHC16, 2015-08-01:
Just submitted second option with a parameter to include/exclude Home as a Rally point.

rmackay9, 2015-08-23:
This is in master now, so closing.  Thanks @KiwiHC16

closed 2015-08-23 by rmackay9