ArduPilot/ardupilot #2472: Copter:New GPS failsafe option:point the nose/tail of the copter towards home and Althold

Akcopter, 2015-06-29:
I want to suggest an addition to the already existing GPS failsafe options of 'Althold' and 'Land', this option can be called 'point the nose/tail of the copter towards home and Althold'
while the name can be different but I think everyone gets my point, the direction to home can be determined using the previously valid GPS position augmented with the estimated velocity data from the previous GPS updates, this new dead reckoned position estimate should be accurate enough to be used for at-least pointing the copter towards home position and in the meantime a GPS failsafe message on the GCS screen should warn and allow the pilot to bring the copter home just by pushing the elevator stick forward or backward.  
why I am suggesting this now is because with more and more people flying with digital HD video links (myself included), the range is much lower, this problem is compounded by the use of high frequencies thus increasing the susceptibility of video signal to obstacles and weather effects, due to this the video link loss is very abrupt which gives no time for the user to react and turn back the aircraft in time, in addition to this if a GPS failsafe condition triggers then RTL wont help the pilot at all and landing wont be the best option if we were flying on water (Assuming the worst), in this case if the Copter enters Althold then because the pilot does not know which way the drone is pointing it would be very difficult to bring the vehicle back without a video link, in this scenario the above option would help a lot, even if the winds are high causing the copter to move, chances are that if the pilot pushes the elevator stick with the copter pointing in the general direction of home, then in almost all cases we will end up with the copter closer to home than before thus increasing chances of regaining the video link and safe recovery, I don't see why this shouldn't be an option in addition to the already existing ones.

robustini, 2015-07-09:
There's "Simple" for this.

Akcopter, 2015-07-11:
Hi Marco.
There has been an entire discussion over this topic on the drones discuss mailing list 
here https://groups.google.com/forum/#!topic/drones-discuss/CGlVAjqky4w
you will find your query answered there.

lthall, 2018-01-11:
Closed due to age and it sounds like it has been discussed and decided on.

closed 2018-01-11 by lthall