ArduPilot/ardupilot #2477: Copter: Compass is not working

ahcorde, 2015-06-30:
Hello @rmackay9 ,

could you please have a look at this issue with APM:Copter:

I'm having troubles with the compass. I have connected an external compass via I2C to the PXF (GPS+Compass - HMC5843). When I connect APM:Copter with APM Planner the compass is not working but APM is receiving the values of the magnetometer as you can see in this graph

[image]

But the log file looks like this:

[image]

This is my list of parameters https://gist.github.com/ahcorde/2d5113aaa794f8f41cbf

I have checked APM:Rover and APM:Plane in both the compass works fine.

ahcorde, 2015-06-30:
I have also checked navio board and the results are the same. With APMrover2 and ArduPlane the compass work fine. But with APMCopter the GSC (APM planner) is receiving the magnetometer's data but the compass' widget is completely stop.  The measurement is only update by the inertial sensors

I don't know if it's something related to EKF o maybe the parameters.

I include people from EMLID to this conversation. @Clouded

vmayoral, 2015-06-30:
Apparently this is something that affects several Linux ports. @dgrat, @lucasdemarchi, @staroselskii, have you guys experienced anything similar?

I'm aware there're some discussions going on regarding the mag. within the MPU9250.

cthill24, 2015-06-30:
This is the issue I was referring to on the DIY forum that we are having Victor.

vmayoral, 2015-06-30:
hi @cthill24, do you mean [gitter](https://gitter.im/diydrones/ardupilot)? If not that i probably missed it.

lucasdemarchi, 2015-06-30:
@vmayoral the only error regarding MPU9250 is the "bad ID for AK8963" for me. Once it pass the configuration, it works and I can fly the copter fine.

Regarding the MPU9250 I was talking to @tridge about it yesterday. I'd say it needs a major refactor, but since he's reluctant with the implications of a refactor I'm now looking on how to fix it with minimal changes to the startup order.... or reviewing a pull request if anyone sends that before me.

vmayoral, 2015-06-30:
@lucasdemarchi, i'm assuming then your compass readings are fine. What hardware are you using? Have you tried any external compass besides the AK8963?

I wonder if reducing the frequency of the sensor https://github.com/diydrones/ardupilot/blob/master/libraries/AP_HAL_Linux/SPIDriver.cpp#L33 would help for us. Will give it a try tomorrow.

lucasdemarchi, 2015-06-30:
> @lucasdemarchi, i'm assuming then your compass readings are fine. What hardware are you using? Have you tried any external compass besides the AK8963?

@vmayoral last time we checked the AK8963 readings, when it works, are fine. However right now our compass is an HMC5883L connected through I2C

> I wonder if reducing the frequency of the sensor https://github.com/diydrones/ardupilot/blob/master/libraries/AP_HAL_Linux/SPIDriver.cpp#L33 would help for us. Will give it a try tomorrow.

Humn... we are running at 15MHz (the spi controller limit in the minnowboard max), not 20Mhz. Not sure it's related, but may be an easy try.

dgrat, 2015-07-01:
I am not entirely sure how the state of the compass implementation is currently.
Nevertheless, for APM just one compass was active. If connecting an external, the internal is ignored (probably by a mux).
Then the trend was to support more than one compass, but on several parts of the code just compass[0] was considered. 
As I said, I don't know how it is now, but I wouldn't wonder if just one is plotted even if other instances are considered at certain points in the program. 
However, I was not testing external compasses recently on linux builds.

ahcorde, 2015-07-01:
Hi @cthill24,
could you provide a link of the discussion? (if it's happening outside gitter)

I have been testing today the PXF and NAVIO, in both cases the results are the same. I'm using a external compass to be sure that the magnetometers measurements are including in the kalman filter, because if I use the internal compass, I don't know if the update came from the inertial sensors or the compass. Therefore:

1) with the board completely static
2) I change the orientation of the external compass but the "virtual compass" (HUD) of APM is static
3) I have checked that the mag.x, mag.y, and mag.z are sent to APM.

I think this is something related to the kalman filter.

Maybe someone could have an idea of what is going on.

rmackay9, 2015-07-02:
Some bits of info that may help:
- COMPASS_EXTERN, 2, 3 parameter should be set for each compass.
- COMPASS_USE, 2, 3 parameters control whether a particular compass is used or not
- COMPASS_PRIMARY decides which compass is used by default, if this compass is not healthy or it's USE=0 ardupilot will use the first compass (starting from 0th) that is both healthy and use=1.
- ardupilot only ever uses one compass, it doesn't mix values from multiple compasses, it also never fails over to a backup compass.
- if the compass information badly disagrees with other info (i.e. gyro) the EKF will stop using the compass and will not try to use it again unless the board is restarted.  a "compass variance" message should appear on the mission planner's HUD (if that's the GCS you're using) in this case.

I've got my head down for the AC3.3 release but when I have a chance I'll try my NAVIO+ board with an external compass.

ahcorde, 2015-07-02:
Thank you for your comments about the parameters, I think it isn't a problem of the parameters because in APMRover2 and APM:Plane the external compass works.

I have deactivated the FS_EKF_CHECK (FS_EKF_CHECK=0) to avoid the problem of "compass variance". But I have never saw this message in my HUD.

I have also tested AC3.3-rc7 and the results are the same.

vmayoral, 2015-07-02:
@lucasdemarchi, @rmackay9  tried reducing the speed of the MPU9250 and together with @staroselskii commits the overall AK8963 "bad ID" aspect seems to improve. However i'm still concerned about how APM takes these mag readings in Copter.

@rmackay9 the title of this issue shouldn't be restricted to the external compass. The same happens with the internal one with the Copter vehicle.

For what we've seen, apparently, in many cases when APM launches the compass points "north" regardless of the orientation of the autopilot.

R-Lefebvre, 2015-07-02:
Victor, if the compass is indicating north when booting, and you're not facing north, the cause is probably because the compass has not booted at all. When there is no compass at all, the copter assumes it is facing north, because it has no other reference.

ivereninov, 2015-07-07:
@rmackay9 @ahcorde @vmayoral  I can confirm this. The test is as follows:

APM:Plane or APM:Rover on master, external compass connected, set to orientation 8 and external.
When you move the external compass with board static, heading in HUD will respond accordingly.

APM:Copter on master, exactly same setup. 
When you move external compass, there is no change in heading at all. 

I have noticed that EKF GBIAS is more strict in Copter, but changing it to the same value as Rover did not change anything.

It is still not clear to me whether it is an issue or EKF is just behaving more strict.

staroselskii, 2015-07-08:
@rmackay9 

I did some testing. I'm going to write down the results of my observations.

I hooked up an external HMC5883 to Navio and launched APM. The data from the compass was transmitted to GCS (APM Planner) where I can verify them in the ```GRAPHS->RAW_IMU.mag```. 

The HUD didn't seem to react on the rotations while holding a vehicle static, though. 

I'm no a Kalman Filter expert but I've tried to find anything that could be related to the issue. In order to do so I added 
```c++
printf("x: %.5f y: %.5f z: %.5f\n", magData.x * 1000.0f, magData.y * 1000.0f, magData.z * 1000.0f); 
```
to ```NavEKF::readMagData```. The output was the same as the data in telemetry. This made me think EKF does fuse magnetometer readings. But the fusion is rather slow (with a ```0.0001f``` gain). 

That's why I decided to give it some time to settle. This yielded some results that one can observe in ```GRAPHS->VFR_HUD->heading```. The heading drifts really slow.. I'm pretty sure the drift is caused my the compass rotation. It might be a coincidence. Extremely misleading, though. This might be the reason it is hard to notice the changes on the HUD screen. 

The only parameter I've changed is ```COMPASS_EXTERNAL```. To sum up, I believe external compass works. I'm not sure if the fusion is right but that's another story.

There's also one issue I'd like to hear a comment on. The EKF compass number in healthiness check is hardcoded. For example, ```_ahrs->get_compass()->healthy(``` ***0*** ```)```.  I'm going to send a PR to discuss this issue separately.

rmackay9, 2015-07-16:
@staroselskii, thanks very much for noticing the EKF issue!  I've got a couple of patches that should go in shortly to master to fix the compass issue so the EKF will use the primary compass instead of the 1st compass.
https://github.com/rmackay9/rmackay9-ardupilot/commit/98621f6fa6922a7bb0ba3af5cc5aa418eca3af27
https://github.com/rmackay9/rmackay9-ardupilot/commit/9b2a444e9b5021414c2446eb94e34d4b70b8a1bf

I tihnk I'll close this issue if that's ok because I think we've determined that the compass is working on linux boards.

closed 2015-07-16 by rmackay9