ArduPilot/ardupilot #2478: Serial 3/4 parameters missing

jeffyromney, 2015-06-30:
In Arducopter v3.2.1 with pixhawk there are some parameters missing

The following Parameters are missing:
SERIAL3_PROTOCOL
SERIAL3_BAUD
SERIAL4_PROTOCOL
SERIAL4_BAUD

They do exist on the latest arduplane (v3.30) as well as I believe on earlier versions of copter

magicrub, 2015-06-30:
I'd this with code you compiled or did you download a binary?

jeffyromney, 2015-07-01:
this is with the automatic firmware upload on mission planner. so a binary.

rmackay9, 2015-07-01:
No,I think these parameters arrived with the introduction of the "Serial Manager".  Before that only Serial2's protocol could be changed in order to support either MAVLink (for telemetry) or FrSky telemetry.

These params are there in the latest beta relase (AC3.3-rc7) and will go out officially with AC3.3.

closed 2015-07-01 by rmackay9

makuser, 2015-09-05:
@rmackay9 what's the reason to close this issue? @jeffyromney is right, those parameters are indeed missing!

magicrub, 2015-09-05:
Right, in the older code they are not there. They are new. If you want
them, upgrade!

makuser, 2015-09-05:
Okay, I did not know that, since 3.2.1 was the first version I ever used.
Upgrading is exactly what I did. I'm now running the current beta version,
but even there I'm having GPS issues. GPS is constantly switching every few
seconds between 3D Fix/no Fix and No GPS. Any idea what it could be?

magicrub, 2015-09-05:
Great question for the copter beta testing group. This issue is closed.

rmackay9, 2015-09-06:
@makuser, yes the AC3.3-beta testing thread is here.
http://diydrones.com/forum/topics/copter-3-3-beta-testing
If you've got dataflash logs with logging-while-disarmed switched on that would be good (set the LOG_BITMASK parameter).
Basically it does that when it's having a hard time figuring out what kind of GPS is being used.