ArduPilot/ardupilot #2479: Copter: PosHold brake can be improved

AndKe, 2015-06-30:
PosHold uses two parameters: 
PHLD_BRAKE_ANGLE  that decides max angle, but not only for braking - it's actually for PosHold - so if set too low, it will drift with wind.   

The leaning sped is set  PHLD_BRAKE_RATE  - if too low, it's stopping very slowly, if it should respond quickly ans is set higher, even at only 12degree/sec  - it causes oscillations (overshoots) when stopping from high speed.

Criuse have a very *perfect* "brake" - it stops aggressively and precisely.  Even Loiter is much better at stopping.

EricXiao1990, 2015-12-04:
Hi, AndKe
You are talking about "Cruise", what's that mean? There's no "Cruise" mode in ardupilot. So It means in auto mode, guided mode or other with automatic flying?

AndKe, 2015-12-04:
Good point, no wonder it works that fine :)  I probably meant Drift , the point was that Poshold have a way of stopping that cannot be tuned to work well in all conditions/speeds.
I will correct the first post not to confuse people, - thanks for finding it.

lthall, 2018-01-12:
Duplicate #4130

amilcarlucas, 2023-09-13:
Loiter has nice break parameters. Use loiter instead

peterbarker, 2024-07-31:
Leonard has reworked braking in POSHOLD since this was raised.

Please test again and reopen this if things are still not right - or just use LOITER as @amilcarlucas suggested!