ArduPilot/ardupilot #2480: Copter: speed set during mission affects RTL

AndKe, 2015-06-30:
This is a safety risk.
Imagine this: you have a mission to circle around a windmill 1km at sea, AP flies there, DO_SET_SPEED sets speed to 1m/s for the photos  - then a failsafe kicks in, or you need it due to loss of video  - now the copter is returning home, at 1m/s.  

mission speed should only be enforced during mission, for RTL and other operations, the configured speed should be used.

magicrub, 2015-06-30:
Sounds reasonable

rmackay9, 2015-07-01:
Yes indeed.  Solo guys have solved this by adding a new RTL_SPEED but perhaps just reverting to the default if the RTL is caused by a failsafe we restore the default speed.

AndKe, 2015-07-02:
Solved by
https://github.com/diydrones/ardupilot/pull/2491

rmackay9, 2016-08-29:
We've added RTL_SPEED to master and this is included in Copter-3.4 (now in beta testing).  Setting this to a non-zero value separates the return speed from the WPNAV_SPEED which is a decent work around for the problem.  This new parameter is mentioned on the Copter wiki on the RTL flight mode page.
http://ardupilot.org/copter/docs/rtl-mode.html

I think I'll close this issue if that's ok.

closed 2016-08-29 by rmackay9

AndKe, 2016-08-29:
Great solution, thank you.