ArduPilot/ardupilot #2482: Calibrations for large airframes

Jman841, 2015-06-30:
With UAV's becoming larger and Ardupilot still one of the most reliable and trustworthy autopilot systems, it is being used in larger and larger airframes. One of the great challenges of this however is the Compass calibration and the 3D Accel calibraton on large airframes.

Is there a way to make a reliable 1D accel calibration feature and compass calibration for large airframes where spinning in circles and putting the airframe on all axis's is impractical?

Thank you

magicrub, 2015-06-30:
For plane I sometimes take the autopilot out and do the accel calibration
externally to the aircraft. No problems there. Also, I've stopping doing
compass calibrations in favor of compass learn which seems to work just
fine.

Jman841, 2015-06-30:
So you just activate Compass Learn before the flight then fly around and it calibrates the compass properly?

That is an option, however, some of our planes have the pixhawks mounted in quite well so it would be difficult to remove to do accel calibrations. Re-instituting an improved 1D accel calibration, or, some other option that doesnt involve moving the aircraft in all positions would be much prefered.

R-Lefebvre, 2015-07-01:
I also often do accel cal with the autopilot outside the airframe, no issue
there.

But the compass is a challenge.  In the past, I have had good results with
"Log Calibration" where you fly around, then MP analyzed the log after you
land, to determine offsets.  In fact, I found this the most reliable in the
past.

But I'm not sure how this would work, or how well Compass Learn would work,
with the EKF, which can crash the copter if it gets a really bad compass
in-flight.

Jman841, 2015-07-01:
Interesting. I definitely think some alternative options would be nice.

Jman841, 2015-09-25:
I think the new on board compass calibration fixes half of this issue. I still think there must be a way to do the accel calibration easier though for large aircraft.

Naterater, 2019-01-07:
The new EKF compass calibration options solve this issue. See https://github.com/ArduPilot/ardupilot/pull/9587

closed 2019-01-07 by Naterater