ArduPilot/ardupilot #2497: sudden impact shutoff

Jgayler, 2015-07-02:
If during flight mode, or during a landing sequence, a sudden deceleration is detected, chances are that the solo has impacted with something. Most likely the ground. There should be an instant shut off that happens when this occurs. otherwise Solo will flip over.

R-Lefebvre, 2015-07-02:
How exactly do we detect this deceleration? Accelerometer spike?  How big and how long?  GPS accel?

magicrub, 2015-07-02:
I'm interested to hear about how this gets done. I have a crude crash
detection for plane that I'm pushing soon and would love to improve it.

rmackay9, 2015-07-03:
I think we should concentrate more on the desired behaviour and raise an issue for that rather than the actual implementation.  We have a crash detector already, in what case does it not perform adequately?

squilter, 2015-07-31:
The crash detector is designed to accomplish this.  It was broken for Solo in a previous release, but it has been fixed now.  Even so, I think the crash detector was never broken for ardupilot master.

closed 2015-07-31 by squilter