ArduPilot/ardupilot #2501: Pixhawk: Start new log file after arming

pepevalbe, 2015-07-03:
If I am not wrong actual behaviour of Pixhawk is that a new log file is created every time we power up.
No matter how many flights you do during a session there will only be one log file.

When I fly a new aircraft I usually do different flights (changing some parameters) and then check the log files. What I do is to disconnect the Pixhakw every time I start a new flight. This way different logs files are created so it is easy for me to review them later at home.

It would be great if Pixhawk generated a new file every time I arm. Do you think this is useful? Do you find any inconvenience in this?

This should be done just in case LogWhenDisarmed bit is not enabled in LOG_BITMASK paramter.

iskess, 2015-07-03:
I have had a similar request for over a year. I want more control over the reset however. Arming won't work well for Planes. See #501.

tridge, 2015-10-22:
Hi Pepe, I think this makes sense. I will assign it to Peter for detailed review

WickedShell, 2015-10-22:
I'm not a huge fan of this personally, it makes it harder to look at what may have gone wrong/happened before the user armed. It is something that seems more like a tooling problem to me. I just don't see how to combine this with logging while disarmed easily.

rmackay9, 2015-10-22:
I agree with @WickedShell that I don't really see an advantage to this behaviour because it will just lead to a huge number of log files.  As a minimum we should ask the wider dev team on drones-discuss@googlegroups.com.

pepevalbe, 2015-10-23:
@WickedShell if logWhenDisarmed bit is not set the user won't see any log before armed with the current code.

When this bit is set everything is logged from startup. My PR doesn't change this behaviour so in that case a single log file would be generated.

peterbarker, 2015-11-25:
A few musings:
 - life gets interesting when the logging isn't entirely under PixHawk control
  - e.g. DataFlash_MAVLink

A few options.

A LOG_CLOSE_ON_DISARM flag.  The current log file would be closed on disarm.  If log-before-arming is set then we would open another log straight away.  Subsequent flights won't contain *all* of the information leading up to a problem in this case, of course; it leads one to wonder what the longest-lived (logged) piece of information in PixHawk's state is.  (mission doesn't count; we emit that at log startup ;-)

@rmackay9 suggests huge numbers of log files would be a problem.  However, having small logs that encompass a single problem is also handy.  Easy, aggressive pruning in post-analysis would make having lots of lots more manageable.  The soon-to-be-announced dronekit-la emits this:

```
find ~/rc/flight-data/peter-x525/logs/ -name \*BIN -print0 | xargs -0 ./dronekit-la -s brief
/home/pbarker/rc/flight-data/peter-x525/logs/2015-11-01/flight1/8.BIN: Score=1244 Crash! Flew
/home/pbarker/rc/flight-data/peter-x525/logs/2015-11-01/flight2/10.BIN: Score=1926 Crash! Flew
/home/pbarker/rc/flight-data/peter-x525/logs/2015-10-18/logs-to-distribute/9.BIN: Score=250
/home/pbarker/rc/flight-data/peter-x525/logs/2015-10-18/logs-to-distribute/13.BIN: Score=2306 Flew
/home/pbarker/rc/flight-data/peter-x525/logs/2015-10-18/logs-to-distribute/7.BIN: Score=601
/home/pbarker/rc/flight-data/peter-x525/logs/2015-10-18/logs-to-distribute/8.BIN: Score=37
/home/pbarker/rc/flight-data/peter-x525/logs/2015-10-18/logs-to-distribute/11.BIN: Score=435
/home/pbarker/rc/flight-data/peter-x525/logs/2015-10-18/logs-to-distribute/12.BIN: Score=116
/home/pbarker/rc/flight-data/peter-x525/logs/2015-10-18/logs-to-distribute/10.BIN: Score=877
/home/pbarker/rc/flight-data/peter-x525/logs/2015-08-01/flight3/25.BIN: Score=1462
/home/pbarker/rc/flight-data/peter-x525/logs/2015-09-20/flight1/13.BIN: Score=949
/home/pbarker/rc/flight-data/peter-x525/logs/2015-09-20/flight1/15.BIN: Score=2933 Flew
/home/pbarker/rc/flight-data/peter-x525/logs/2015-09-20/flight1/16.BIN: Score=358 Flew
/home/pbarker/rc/flight-data/peter-x525/logs/2015-09-20/flight1/14.BIN: Score=1474 Flew
/home/pbarker/rc/flight-data/peter-x525/logs/2015-09-20/flight1/.BIN: Score=80 NoVehicle
/home/pbarker/rc/flight-data/peter-x525/logs/2015-08-30/24.BIN: Score=1061 Flew
/home/pbarker/rc/flight-data/peter-x525/logs/2015-08-30/22.BIN: Score=418 Flew
/home/pbarker/rc/flight-data/peter-x525/logs/2015-08-30/20.BIN: Score=622 Flew
/home/pbarker/rc/flight-data/peter-x525/logs/2015-08-30/21.BIN: Score=1355 Flew
```
Anything without "Flew" can usually be immediately discarded.

Having the parameter also allows you to continue to have a single, continuous log (from boot or while armed...)

pepevalbe, 2015-12-07:
Hello @peterbarker. I forgot to reply when you commented.
There is already a Pull Request for this feature: #3067 

We finally decided to use a new parameter to specify the logging behaviour:
 * log always
 * log only when armed
 * log only when armed and reopen file

magicrub, 2015-12-08:
Log only when armed is Master in the log bitmask 1<<16 I think.

pepevalbe, 2015-12-08:
Yes you are right, but Randy suggested moving the logging-while-disarmed bit to this new parameter.

OXINARF, 2016-12-17:
This is now in master with the parameter LOG_FILE_DSRMROT.

closed 2016-12-17 by OXINARF

yourobmob, 2019-07-15:
i ' have a problem , how can we do the log all the time , a log with a disarming ? t

peterbarker, 2019-07-16:
@yourobmob set LOG_DISARMED to 1

exilednova, 2019-07-18:
Can someone please tell me how can I review log files while armed? Or If I can possibly create a new log file also while armed? Thanks in advance

rmackay9, 2019-07-19:
@exilednova, reviewing a log file is normally a feature of the ground station.  For example MP's Flight Data screen has a "DataFlash Logs' tab on the bottom left and the "Review a Log" button will work regardless of the vehicle's state.

Log files are normally started when the vehicle is armed (although it can be configured to be at startup).  We don't support writing two log files at once so the best you can do is write to the current log file that is already being written to.

In case you're looking to write new log file messages, we've got some [advice on the developer wiki](http://ardupilot.org/dev/docs/code-overview-adding-a-new-log-message.html)