ArduPilot/ardupilot #2506: Plane: Altitude estimation recalibration

Jman841, 2015-07-04:
On longer flights (30+ min) and on days when the weather is changing quite a bit, we have issues with the altitude estimation drifting quite a bit. The main issue besides the altitude being a little off in the air, is the automatic landing. Yesterday after a 45min flight, one plane landed but the altitude was still predicted to be at 5 meters above the ground so when it landed it turned the motors on thinking it was still flying. 

Is there a way to recalibrate the barometric sensor in flight to account for changes in barometric pressure? 

Maybe by user input or it can do a sanity check with another sensor. First, if using A laser distance sensor, as soon as it has readings from that maybe it can automatically recalibrate the barometric sensor compared to that data?

Maybe add a second barometric pressure sensor that monitors changes in the natural atmosphere when the plane is not changing altitude for a period of time?

Maybe a sanity check with the GPS altitude estimates? I'm not sure how inaccurate the GPS estimates are to work as a recalibration check, but maybe another option?

Our current fleet of planes includes some electrics with pixhawks that can fly over 2 hours and we expect to see up to 2 times that in flight time soon. These altitude estimates seem to get worse as flight time goes on and is becoming concerning.

Thank you for all the hard work you guys do making this one of the best autopilot systems available.

magicrub, 2015-07-04:
Yes yes and yes.
Lidar is the easy solution, I suggest you look into the lidar lite.

Jman841, 2015-07-04:
I have a couple but have not installed them yet. I guess that's the next step.

So when you say yes yes and yes, are all 3 of those things already implemented?

Is there any way without the LIDAR to improve the altitude drift over time?

ThorstenDP, 2015-07-04:
Have a look at: https://github.com/diydrones/ardupilot/pull/2310
This should solve the issue. LIDAR is perfect for landing but not for handling general altitude drift.

magicrub, 2015-07-04:
Closing because this is a solved issue. Better handled via drones-discuss or the forums.

Yes (sort of), GPS can be used for altitude in older code, but was removed in the last couple releases due to it's inconsistency world-wide. In the USA we have WAAS-SBAS which gives us accuracy slightly better than long-flight baro does but, for example, in Australia where there is no SBAS system you can drift several tens of meters. per Thorsten's post of issue 2310, that feature is coming back.

Yes, Differential Baro is what you are describing via using the GCS with a ground referenced baro. During a flight you can set the "ALT_OFFSET" param to give it an arbitrary offset to account for baro drift. If it's not already done, there is at least talk of doing this in Antenna Tracker automatically.
See http://plane.ardupilot.com/wiki/arduplane-parameters/#altitude_offset_arduplanealt_offset

Yes, Lidar is the easiest and most robust (when it works) solution.  See these two pages for info on that
http://plane.ardupilot.com/wiki/flying/automatic-landing/
http://plane.ardupilot.com/wiki/common-optional-hardware/common-rangefinder-landingpage/

closed 2015-07-04 by magicrub

Jman841, 2015-07-04:
Ok thank you,

So if you have a differential baro hooked to your computer you can update the offsets? Any idea where to get one of these or how to get the proper number offsets while it is in the air?

Is there anyway to display the current GPS Altitude reading during the flight to compare it with the Baro altitude and see if it is accurate for my location?

WickedShell, 2015-07-04:
I was about to tell you that you can, but I actually don't see it offered in mission planner. If you have access to the packet that gives GPS_RAW_INT its reported there. (So its easily available on MAVProxy, I'm not sure about any of the other GCS options).

Having used the ALT_OFFSET to correct for some fairly large errors over 3 hours you need to be careful how fast you change it. Large changes (~>5 meters) can (and have for me) caused the EKF to panic for a moment and throw a warning. Its better to slowly transition the ALT_OFFSET rather then do a large change. If you need to accomplish a large change then slowly drift it. Also realize that this can cause your logs to look a little werid for photos. (Sudden discontinuities in the reported height of the photo around where the offset was applied if I recall correctly. Whatever it was the problem was easily handled with GCPs).

WickedShell, 2015-07-04:
Also an option for the updates (not a great one mind you is to do a low pass flyby at a known altitude (IE the safety pilot flew the plane over the ground about 2 meters up, and I pulled the correction number from the reported height on the GCS). This is a poor way of doing it though as it requires leaving the mission to do the flyover, has some guesswork, requires pilot skill and takes time away in flight. (And its really inconvenient, the second sensor on the GCS approach is far superior)

Jman841, 2015-07-04:
Thank you for the info. Could we get that added to mission planner to it can be used for adjusting offsets. And is it possible to add a parameter to add a second ground baro (Still confused what to use to get this info? A second pixhawk?) and if set to "On" it will automatically correct for the offsets throughout the flight?

Should I open a new issue for these specific requests?

WickedShell, 2015-07-04:
I've used a second APM in the past (pixhawk would work, but since I had a stack of APM's that weren't seeing flight use anymore...). Ideally it would be a single sensor that was connected (IE a USB dongle that costs less/takes less space then an additional FC).

This is really the realm of a GCS implementation to add the automated updating, I'd open an issue with the respective GCS's. Its worth pointing out that this approach only works well if your GCS is fairly stationary (IE if you walked your second baro into a canyon the plane would follow you down). It's worth checking with @priseborough if there is a better approach to updating the baro long term though. There are a lot of issues floating around to handle that. People's desired solution tends to change depending on who you talk to. (IE I want to be able to update the adjusted MSL pressure in inches of mercury or bar's and use that to pull relative trends from nearby weather stations similar to how manned aviation handles it).

As far as MP showing the gps altitude you'd have to ask Michael Oborne, it's possible I missed it in the list (stranger things have happened).

Jman841, 2015-07-04:
OK, thank you. I am open to any solution that works best, but I do think it is an area for improvement.

I will open the issue with Mission Planner to display GPS altitude (Although if GPS Altitude is reporting differently, how do I adjust the offset between the two? Is it still the same "ALT_OFFSET" parameter?)

I believe your pull request is already trying to address the baro offsets so I won't open another issue here, Although I would like to see a gradual offset implementation automatically happy if the ground baro sensor is being used.

Also, from what I read it is not possible to use the GPS altitude over Baro altitude anymore? We are about to being experimenting with RTK GPS and I believe that would be considerably more accurate than the Baro. How would I implement that with the current firmware?

Sorry for all the questions, should I move this discussion somewhere else?

WickedShell, 2015-07-04:
You compare the offset between GPS altitude and the current altitude as reported by the aircraft. Just to note if you ever have poor performance GPS that fluctuates on the aircraft this is probably not the way to go. (That said if you have RTK then it should be fine).

I don't have a pull request in. I'm not sure if there are any around for it. (If there are and I've missed it then please let me know so I can try it).

The GPS altitude over baro altitude is what is in pull request #2310 is for. It just hasn't been pulled in yet.

magicrub, 2015-07-04:
As mentioned earlier, GPS based altitude is not currently supported but
will be soon. The drive for that is RTK.

Please open an issue in the mission planner github regarding viewing the
additional altitude options of they are mixing.

antenna tracker is a compile option for ardupilot to make your apm/pixhawk
control a pan tilt directional antenna. I believe it will use its baro to
adjust the target aircraft's alt offset.

I use lidar and it works for me. You have them on hand, I'll suggest you
give it a try.

We need an issue opened to smooth out alt-offset writes so step inputs
don't trip up the ekf.

WickedShell, 2015-07-04:
Actually talking to Michael it was kinda obvious that maybe it makes more sense to do that update on the autopilot since it could compare the two values directly and do the math, rather then on the GCS (if you are using a GPS). I just reached for the ground station solution as its what I've done in the past. (Although a baro solution would still need GCS support)