ArduPilot/ardupilot #2507: Misleading yaw controller easily cause crashes

mah92, 2015-07-04:
AC3.2 allows the desired yaw to be dragged around by the actual yaw.
Even if this feature is developed to make the flight safer, it indeed is a very dangerous solution. We just had a crash because the gust rotated our yaw and then desired yaw obeyed too, which caused our pilot to loose where the aircraft was pointing.
By reading documentation, every pilot assumes that the aircraft will not rotate permanently if he does not command to, and especially when the aircraft goes farther out of sight, this becomes more important.
Another note is that it was not just gust. Any other source of yaw moment like change of collective without feed forward caused undesired temporary yaw that by changing desired yaw this caused permanent yaw and could lead to catastrophe.
Overall, adding this feature is like that you **disabled yaw control**, and **even worse**.

R-Lefebvre, 2015-07-04:
In the second graph, was the copter armed when you did that test?

mah92, 2015-07-09:
Yes, if not armed, it would not save any log. Notice that arming check is not activated in stabilize mode. Please take this problem serious, its more like a bug than a feature.

silviorevelli, 2015-07-16:
Need a compromise; both solutions make sense.

mah92, 2015-07-22:
When copter is far, this solution is dangerous, specially in a windy day that pilot commands lean angle to resist wind, pilots first mistake in determining yaw will be his last mistake(as occurred to us). At least it should have an option determining delay.

silviorevelli, 2015-07-22:
Althold, PosHold, RTL, Auto prevent yaw dragging.

jeffyromney, 2015-07-23:
what are the benefits of this feature?  It must have been added for a reason, but I can't think of a situation where I would prefer my yaw to change without me telling it to.

silviorevelli, 2015-07-23:
I guess that the answer is in the meaning of "Stabilize".
In Stabilize your Alt can change without you telling it to, however you can correct manually.
In Stabilize your Pos can change without you telling it to, however you can correct manually.
Similarly, also Yaw can change without you telling it to, however you can correct manually.
If you want an "auto correction" of each of them (Alt, Pos, Yaw) you should rely on specific sensors (respectively barometer, GPS and compass). Stabilize relies only on accell/gyro, which is very robust to external interferences; it does make sense.

rmackay9, 2015-07-24:
The idea behind this feature is it prevents a sudden backlash if the vehicle is disturbed or has been prevented from reaching it's goal by some physical obstacle.  It is consistent with the altitude and position controllers.  So for example, if you drag a vehicle from it's current loiter position about 10m away and then let go of it, it won't return to it's original spot but rather some other point in between it's current and original spot.
Roll and Pitch controllers act the same but you don't see it because the pilot's sticks are constantly inputting an absolute angle command (yaw/heading input is a rate control).
Maybe we could widen the angle error at which the target is adjusted.

silviorevelli, 2015-07-24:
So, the system uses the compass in two cases:
- "Condition yaw", which sets the heading only once
- "Face to next WP" which costantly sets the heading
In all other cases the pilot controls the yaw rate. 

I suggest to keep this feature in stabilize mode, because stabilize mode should rely only on accels/gyro, while alt, pos, heading are manually controlled through throttle, lean angle, yaw rate. Maybe the vehicle should try to keep the current heading using only gyro but not compass.

If you fly near magnetic fields, the compass could be totally wrong/disturbed, causing chrashes. 

@rmackay9 I wouldn't rely on angle error (compass) but only on rate/accel angle error (gyro). Obviously I don't have the big picture as you, so maybe I'm missing something.

lthall, 2015-07-24:
This looks like you are trying to make the autopilot compensate for poor pilot skills or copter setup.

How did the copter rotate in yaw. I have only seen this on my high speed quad at over 100km/h?

How did this lead to a crash? I can loose yaw orientation at night, 200m away, and regain orientation and fly the copter back to myself using nothing but the led on the pixhawk to see the copter.

The purpose of limiting the yaw error such that it will drag the desired heading is to ensure that aggressive maneuvers or an external impact will not result in the copter appearing to rotate out of control for a period of time afterwards. This ensures the pilot has complete control as soon as possible after any incident or impact that rotates the copter.

The other situation this limited yaw error is designed to handle is where the copter  rotated as it touches down or as the throttle is increased on take off. This can happen if the legs of the copter catch on uneven grass for example. This can result in the copter having a 90 degree (for example) heading error if the error isn't limited. Without this feature, as soon as the copter takes off it would rotate to its original heading. This may result in the pilot becoming disorientated very close to themselves or others. The obvious danger here is a copter in the face.

As I think has already been pointed out. The copter does return to its desired heading in autonomous modes. However, there would need to be a very clear argument made to justify doing this in any manual mode.

jeffyromney, 2015-07-24:
I see the reason now. thanks. I like @rmackay9 's Idea:
"Maybe we could widen the angle error at which the target is adjusted."

I think that would be a good compromise.

silviorevelli, 2015-07-24:
Can someone point me to the lines of code related to this?

rmackay9, 2015-07-25:
@silviorevelli, here are the important bits here:
https://github.com/diydrones/ardupilot/blob/master/libraries/AC_AttitudeControl/AC_AttitudeControl.cpp#L497
https://github.com/diydrones/ardupilot/blob/master/libraries/AC_AttitudeControl/AC_AttitudeControl.h#L36

It looks like the angular limit is set to 10degrees so changing the second link's definition to be say 3000 would make it 30degrees.  One warning is that at some point we found that if we allowed the yaw error to be too big it would lead to a wobble in the roll or pitch.  This was related to the attitude controller using euler angles (we plan to switch it to use quaternion's in AC3.4) and the relatively weak yaw control on most copters.

mah92, 2015-08-15:
Hi all, sorry about delay. 
@silviorevelli The problem is that actually we had the crash in AltHold not stabilize, dragging was not present in stabilize mode! documentation says that stabilize & AltHold  are just different in altitude hold controller but its wrong.
@silviorevelli In stabilize mode all sensors except baro and gps can be used. why not to use magnetometer? even if you don't use magnetometer, as I tested integrating z gyro would give correct yaw +-5 deg even after 1 minutes.
@lthall We did not have a poor setup, it was a windy day. Your leds on pixhawk do not give any benefit in daylight!
overall, I prefer disabling yaw integrator over this complex solution, and I think providing an option to choose max angle error would be best. Even you could just set a saturation limit and keep integrator but not use this unpredictable solution. 
thanks.

weihli, 2016-01-05:
I fly my quad copter(AC3.2.1) well, but after replacing battery, I take off quad with stabilize mode,  the quad rotate around yaw automatically;  there's no gusty; I saw the log, actual yaw drag the desired yaw  like @mah92  mentioned. After several hours, I fly again, but everything goes well; i doubt the compass  interfered; so the DCM get wrong yaw angle after drift_correction_yaw with compass's data.

lthall, 2018-01-11:
This is a combination of poor aircraft setup and pilot error. The code is behaving as designed.

closed 2018-01-11 by lthall