ArduPilot/ardupilot #2508: Altitude Estimation Adjustment integration

Jman841, 2015-07-05:
As discussed here :https://github.com/diydrones/ardupilot/issues/2506#issuecomment-118562406

When adjusting altitude estimates based off GPS altitude, Ground station Barometer, or other method, the quick jump of a large offset (Greater than 5m) causes the EKF to go crazy and can mess with the logging for imagery.

Would it be possible to implement a change in Altitude estimation adjustments in a smoother way that does not cause the system to have errors?

Thank you

magicrub, 2015-07-05:
Thanks @Jman841 

After looking into this it seems we have two altitude offset options, but only one is "hooked up". In plane there's "ALT_OFFSET" which works as we expect it to, copter doe snot have it. Then there's "GND_ALT_OFFSET" which is in the baro library for all platforms to utilize which is the better way to deal with this since copter and plane both suffer from the same problem although it's more critical for planes. However, it seems this value is stored but unused. Odd that someone implemented it half-way like that like a future-proofing place holder.

WickedShell, 2015-07-05:
IIRC I used GND_ALT_OFFSET in Plane 3.2 but I could be wrong.

As far as the log I can't recall if it adjusted the altitude in the tags or not. Whatever the problem was it was simply that in the image processing software the photo was misidentified in terms of altitude and required me to compensate by manual adjust the altitude up or down. (Or just let GCPs handle it). I unfortunately can't recall which it was and am no longer next to the data set for the next week or so.

magicrub, 2015-07-05:
Pull Request pending: https://github.com/diydrones/ardupilot/pull/2509

rmackay9, 2015-07-05:
As discussed with Tom, I think the general plan going forward (discussed only with PaulR and Tridge) is that we shouldn't smooth things but rather propagate a flag (or something) through the system when a sensor or estimate's values jump.  So the consumer should some how be made aware of the jump so that it can deal with it quickly and effectively.

magicrub, 2015-07-05:
@CraigElder can you add this to the next dev meeting topic?

Jman841, 2015-07-15:
A little off of this issue but not a big enough deal to open a new one.

Would it be possible to have the altitude calibration numbers added to Tower or other smart phone GCS? Since almost all smart phones have a baro in them now it would be as simple as reading the information from your phone in the app and adjusting it in your flight. 

The issue of changing offsets during flight still persists, but, this could be a simple way to get the correct new offsets after long flights to help with altitude accuracy.

magicrub, 2015-07-15:
May be a good use for alt_baro

magicrub, 2015-07-15:
I mean GND_ALT_OFFSET

Jman841, 2015-07-15:
I agree, I brought the issue up in Tower, so hopefully it's something they can add. However, the issue of making large adjustments to GND_ALT_OFFSET still exists.

Any updates from the dev meeting on the integration of large offset changes?

magicrub, 2015-07-15:
There was talk of setting a flag for the ekf to blend it in.

Jman841, 2015-07-16:
I was looking into this more today and playing around with a spare pixhawk. I am a bit confused. Is ALT_OFFSET the only parameter currently implemented or is GND_ALT_OFFSET also working in current version APM 3.3.0?

When adjusting ALT_OFFSET I can see the ALT_ERROR parameter under Status on mission planner change instantly. When I adjust GND_ALT_OFFSET I don't see any changes. However, I am testing this on the ground. I am still a bit confused as to the difference between the two parameters and which baro parameters are used for comparison. The Wiki and description does not seem to explain the differences of the two well.

magicrub, 2015-08-01:
Since the EKF flag thing probably won't happen for awhile, here's something that will. Simply lowpassing the inputs so there's no step. It's a super slow filter, several seconds.
https://github.com/diydrones/ardupilot/pull/2635

R-Lefebvre, 2015-12-15:
Just checking, is this issue still open?  Any chance it got fixed in EKF2?

rmackay9, 2015-12-15:
I'm not sure if we're all talking about the same thing but if we're discussing a position reset being recognised by the control loops (to avoid a jerk), that hasn't happened yet.

WickedShell, 2016-02-09:
@magicrub since #3568 is in can this be closed?

Naterater, 2019-01-07:
@magicrub, can you please update this?

closed 2021-07-26 by Naterater