ArduPilot/ardupilot #2521: RC Input in BeagleBone Black

TempleWaterfOWLS, 2015-07-06:
I have a BeagleBone Black running APMrover2 (on a boat) and I'm having problems getting RC control to work. 

We have a radio, receiver, and 3DR PPM encoder. We scoped the output of the PPM and it changes as we move the RC radio. The signal from the PPM encoder is attached to P8_15. With the radio turned on and running APMrover2.elf the radio isn't found when doing setup/radio in CLI mode. Thoughts?

Here is the list of overlays:
0: 54:PF---
1: 55:PF---
2: 56:PF---
3: 57:PF---
4: ff:P-O-L Bone-LT-eMMC-2G,00A0,Texas Instrument,BB-BONE-EMMC-2G
5: ff:P-O-- Bone-Black-HDMI,00A0,Texas Instrument,BB-BONELT-HDMI
6: ff:P-O-- Bone-Black-HDMIN,00A0,Texas Instrument,BB-BONELT-HDMIN
7: ff:P-O-L Override Board Name,00A0,Override Manuf,BB-UART5
8: ff:P-O-L Override Board Name,00A0,Override Manuf,BB-BONE-PRU-05
9: ff:P-O-L Override Board Name,00A0,Override Manuf,BB-SPI0-PXF-01
10: ff:P-O-L Override Board Name,00A0,Override Manuf,BB-SPI1-PXF-01
11: ff:P-O-L Override Board Name,00A0,Override Manuf,am33xx_pwm
12: ff:P-O-L Override Board Name,00A0,Override Manuf,bone_pwm_P8_36

magicrub, 2015-07-06:
@mirkix (BBBmini) or @vmayoral (BBB PXF), are either of these under your domain and/or able to comment on it?

TempleWaterfOWLS, 2015-07-07:
Tried running the example code in Tools/Linux*/rcinpru.

Here is the output:
root@beaglebone:~/ardupilot/Tools/Linux_HAL_Essentials/rcinpru# make
clpru -v3 -s -al -O3 --c99 --gcc --printf_support=minimal --symdebug:none -c rcinpru0.c -ea=.s
"rcinpru0.c", line 7: fatal error: could not open source file "stdio.h" (no directories in search list)
1 catastrophic error detected in the compilation of "rcinpru0.c".
Compilation terminated.

Not sure if that helps or not.

TempleWaterfOWLS, 2015-07-07:
More information:

Running rcin_test.c from Tools/Linux_HAL_Essentials/test_codes. Doesn't give any errors but also doesn't output any information even when RC is connected to PPM (the blue light is solid).

The PPM encoder is connected to header 8.15 on beaglebone black.

TempleWaterfOWLS, 2015-07-07:
Looking at Tools/Linux_HAL_Essentials/startup.sh and I've never run $./startup.sh reload.

Tried running and it says:

root@beaglebone:~/ardupilot/Tools/Linux_HAL_Essentials# ./startup.sh reload
Loading Firmware...
./startup.sh: line 20: /sys/devices/ocp.3/4a300000.prurproc/load: No such file or directory

Seems important. Thoughts?

mirkix, 2015-07-07:
@TempleWaterfOWLS Can you tell us which operating system do you use on your BBB? It seems it is not possible to load the PRU firmware.

TempleWaterfOWLS, 2015-07-07:
I am running Debian.

Output of /etc/*-release:
PRETTY_NAME="Debian GNU/Linux 7 (wheezy)"
NAME="Debian GNU/Linux"
VERSION_ID="7"
VERSION="7 (wheezy)"
ID=debian

vmayoral, 2015-07-07:
How about the kernel? Can you print `uname -a`? I'd guess you didn't include the PRU patches when you compiled the kernel.

TempleWaterfOWLS, 2015-07-07:
Output of /proc/version:
Linux version 3.8.13-bone47 (root@imx6q-wandboard-2gb-0) (gcc version 4.6.3 (Debian 4.6.3-14) ) #1 SMP Fri Apr 11 01:36:09 UTC 2014

Output of uname -a:
Linux beaglebone 3.8.13-bone47 #1 SMP Fri Apr 11 01:36:09 UTC 2014 armv7l GNU/Linux

vmayoral, 2015-07-07:
@TempleWaterfOWLS seems like you are using one of the old Debian images from BeagleBoard. Feel free to fetch any of ours https://github.com/erlerobot/bbb-images/blob/master/README.md. @mirkix has also done a good job putting together a Debian FS with the latest changes.

TempleWaterfOWLS, 2015-07-07:
@vmayoral Thanks for the update I'll try re imaging the BeagleBone.

Is there a way to just add the needed module to the Kernel without re imaging?

TempleWaterfOWLS, 2015-07-08:
@vmayoral I got the erle-debian-flash-25-4-15.img.gz image onto my BeagleBone but I can't ssh into it. It is no longer at 192.168.7.2 when USB connected. Does it have a different default address?

vmayoral, 2015-07-08:
no, it's that one but i've disabled the DHCP server on the Ethernet-over-USB interface so you should make sure that your network interface has the `192.168.7.1` (or something in the same subnet) assigned.

TempleWaterfOWLS, 2015-07-08:
So my computer's IP was on a different subnet. Just went in to make it static to 192.168.7.1 and I can now ssh into the BeagleBone using 192.168.7.2.

Will let you know if the rc control works.

TempleWaterfOWLS, 2015-07-08:
Got the RC Control working (at least through the CLI which is better than I've had before).

Thanks everyone!

vmayoral, 2015-07-14:
@magicrub you can probably close this issue now.

rmackay9, 2015-07-14:
Ok great, closing!

closed 2015-07-14 by rmackay9