ArduPilot/ardupilot #2524: Bug: APM Plane 3.3 sets flaps in flaperon as spoilers

marcmerlin, 2015-07-07:
I have flaperon_output set to 1 and my ailerons are channel 5 and 6
In CRUISE mode, if I tilt my plane, both the ruddervators and the ailerons go in the right direction.
I then have FLAP_IN_CHANNEL set to 7 and when I turn the knob from 900 to 2000 the ailerons go up, acting as spoilers, no flaps like I need.
There is no way for me to invert this.

The docs say 
"When enabled, the percentage of flaps is taken as the percentage travel from the TRIM value of the channel to the MIN value of the channel. A value above the TRIM values will give inverse flaps (spoilers)."
It's not clear what "above trim values" (with an 's') means. Above RC7_TRIM? If so, that's not true. The RC value going from 800 to 2000 only makes the flaps go in one direction (spoilers).

Fixing the issue that RC7 does not make flaps go in either direction but only from 0 to 100%, would easily fix my problem that I'm only getting spoilers and not flaps.

marcmerlin, 2015-07-07:
Obviously is I change FLAPERON_OUTPUT to 4, then flaps work as flaps, but the plane would then turn in the wrong direction and crash if I used that.

magicrub, 2015-07-07:
Sounds like a horrible bug. Do any developers out there have a flaps setup
available to confirm and/or fix this?

marcmerlin, 2015-07-07:
Details I forgot: this is a new plane, I have never setup FLAPERON_OUTPUT before because I was doing flaps mixing on my controller (which only worked in MANUAL mode of course). 
So I don't know if it would have worked better with AC3.2 or older.

Last but not least, I don't know if auto flaps work or not, I'm not sure how to test them on the ground.
Maybe those would work and it's only the manual flaps control that is reversed or doesn't go from -100% to 100% and only goes from -100% to 0% like it should (as per the documentation at least)

magicrub, 2015-07-07:
I think this is supposed to work without transmitter mixing. The mixing is
handled by the autopilot. Perhaps that's what is screwing things up? Have
you checked the wiki page for flaps?

marcmerlin, 2015-07-07:
I don't have any mixing on the transmitter, it sends aileron on channel one, and flaps on channel 7.
AC3.3 generates the signal for RC5 and RC6 just fine (no signal from the TX on those channels), the only problem is that the flaps input makes them go into spoilers instead of flaps.

magicrub, 2015-07-07:
".. I was doing flaps mixing on my controller [in Manual]"
Oh sorry, I read this wrong. Thought you meant during auto.

marcmerlin, 2015-07-08:
Hi @tridge do you happen to be using this? Does it work for you? (flaperons being flaps and not spoilers?)  Thanks, Marc

KIrill-ka, 2015-07-14:
> Obviously if I change FLAPERON_OUTPUT to 4, then flaps work as flaps, but the plane would then turn in the wrong direction and crash if I used that.

I think you can solve your problem by using proper combination of following controls:
- CH1_REV
- by exchanging RC5_FUNCTION and RC6_FUNCTION
- aileron reverse in TX
- FLAPERON_OUTPUT

> "When enabled, the percentage of flaps is taken as the percentage travel from the TRIM value of the channel to the MIN value of the channel. A value above the TRIM values will give inverse flaps (spoilers)."

That's incorrect. It doesn't use TRIM value, just the percentage from min to max, so you can't invert it that way.

marcmerlin, 2015-07-15:
Thanks, I'll try that when I get home. It'll take a few days.

marcmerlin, 2015-08-03:
@KIrill-ka . I got to try this today, sorry for the delay.
1) there is no CH1_REV, if you meant RC1_REV that seems to do nothing since I had no servo plugged into CH1
2) If I change RC5 and RC6 FUNCTION, then I have to change FLAPERON_OUTPUT from 1 to 4 for Cruise to move the flaps in the right directions, and one I do that, my flaps input control, again, moves the flaps up and makes them spoilers instead of flaps

The only way to have the flaps go down with my switch is to have FLAPERON_OUTPUT the wrong way, and then it works for flaps from my switch and manual ailerons, but if I switch o CRUISE, they're reversed and it would be a disaster in flight of course.

I do need something like that CH1_REV you mention, but there is no such setting.
ELEVON_CH1_REV sounds like what I'd want, except for flaps.

Am I still missing something, or is there a bad bug here?
Thanks

marcmerlin, 2015-08-04:
@tridge : could I appeal to you to have a look at this and confirm if I'm doing something wrong, missing a badly/non documented option, or if there is a real problem here?
I'm sure you must be using flaperons on your planes, right?
Thanks.

KIrill-ka, 2015-08-04:
> there is no CH1_REV, if you meant RC1_REV that seems to do nothing since I had no servo plugged into CH1
RC1_REV, my mistake. It doesn't matter that you don't have ch1 servo connected. Flaperon function mixes computed ch1 output with flaps, so ch1 reverse should work. Note that REV parameters take 1 and -1, not 0.

marcmerlin, 2015-08-04:
@KIrill-ka I did try RC1_REV yesterday and it didn't reverse what CRUISE did, but I just tried it again now and it is working.
So I then had to reverse the CH1/aile on my TX and things seem ok now.

Mmmh, so this was all very non obvious. I'm happy to write a page on flaperons on the wiki. Do you have a way to give me access, or should I just write a blog post and publicize it?

magicrub, 2015-08-04:
A wiki page is best, more legitimate and reaches more people who are
searching for the right config. We'll need to ping Craig who will ping
Hamish.

gmorph, 2015-08-04:
@marcmerlin - can you please attach a dataflash log (downloaded from your ardupilot) whilst your plane is armed (to ensure we get logging) and could you please methodically move the RC input sticks on your transmitter one at a time (not throttle) for a period of 5 seconds each one.  So, elevator down for 5 secs, elevator up for 5 secs, etc .  This will help us diagnose the problem.
Thanks, Grant.

marcmerlin, 2015-08-04:
@gmorph the problem is fixed now as per my comment from this morning. The issue is total lack of documentation for a setup that isn't obvious in any way.
I'm going to write something up to save others the trouble I went through

gmorph, 2015-08-05:
Glad its working.  We are always keen to get improved documentation.
Thanks, Grant.

closed 2015-08-05 by magicrub

hamishwillee, 2015-08-05:
@marcmerlin Fantastic. I've created a docs placeholder for this [here](https://github.com/diydrones/ardupilot-wiki-issue-tracker/issues/120) along with information about getting started.

marcmerlin, 2015-08-05:
If anyone finds this bug with google later, I wrote up instructions here: http://marc.merlins.org/perso/rc/post_2015-08-04_Using-Flaperons-With-Ardupilot.html